x-station/xstation-issues #63: Question regarding pin79

MegaPramuk, 2021-02-01:
Technically this isn't really an issue about xStation but I couldn't really find anything online and it is at least related.
Does pin 79 of the CXD2545Q have a via pad or any other point where it should have continuity with?
Thing is, I was installing my xStation and everything is soldered correctly as far as I could test it but I had a mishap happen to me while lifting the pins.
I lifted pin 79 and tried to solder it back onto the point where it broke off cleanly.
Shaving off the IC a bit to expose the pin did the trick to solder onto but the place where the pin was soldered to is too narrow, since I've tried it several times (got bridges with neighbouring pins) and now the top most layer is peeling off, leaving no way to apply solder.

Thanks in advance.

ramapcsx2, 2021-02-01:
Pin 79 is an input for a MUTE signal. It may not even be necessary.

ryanm101, 2021-02-03:
you could scrape some of the trace that the pad connects to to remove the silkscreen, then run a bodge wire (very fine magnet wire) from the IC to the exposed copper of the trace. if you are getting bridging you likely have a) too much solder and b) too little flux (No such thing as too much flux)

closed 2021-02-21 by ramapcsx2