x-station/xstation-issues #64: SD Card Corruption Issue? 2 Different SD card mfgs, 2 different XStation installs

bleggett, 2021-02-02:
I've successfully installed 2 xStations that work wonderfully, thanks!

But I've noticed a strange issue that I at first wrote off as a bad SD card until it happened again.

The issue is this - after installing an xStation I 

- format a new 128GB microSD card using the recommended SD Card Formatter
- create the `00xstation` folder with 1.0.5
- slap a disc image on there
- boot up the xStation, upgrade the firmware to 1.0.5, and load a game

Shortly after this, the microSD card (while remaining readable by the xStation) becomes completely unrecognized by any PC or Mac I try it with, no matter what port or SD card adapter I use. And when I say unrecognized, I mean it doesn't even show up as a corrupted volume - the microSD is completely undetected by any other device.

I can plug this damaged microSD into the xStation, which reads it fine, but after this point the microSD is basically unusable - no tool I've tried can detect it, reformat it, anything.

The first time this happened, I found some docs here suggesting Samsung brand microSD cards should be avoided, assumed it was a faulty microSD, and bought a Sandisk 128GB which works fine in my personal xStation.

For the next install I did, I got a Sandisk 128GB microSD from the start - and the **exact same thing** happened to the friend I shipped the modded console to - microSD is read by xStation just fine, but from then on is undetectable by any other device.

Could the xStation be causing some kind of really basic disc corruption here? I would note that the one constant here is that the cards that fail like this are the ones I use for the initial firmware upgrade, perhaps there's a bug there?

ramapcsx2, 2021-02-02:
Hey,
I have not encountered such a problem over the year of development, nor with any of the dozens of cards tested.

My best guess is that it has to do with your micro SD card reader. If you can, try a different one :)

bleggett, 2021-02-02:
> Hey,
> I have not encountered such a problem over the year of development, nor with any of the dozens of cards tested.
> 
> My best guess is that it has to do with your micro SD card reader. If you can, try a different one :)

I've tried it with 3 different readers, and the person I sent the second console to tried it with their own readers.

It's possible I got back to back faulty SD cards from 2 different manufacturers, but that seems unlikely.

ramapcsx2, 2021-02-03:
It's the first time I hear of such an issue.
Something major like this would have shown up in the beta tests, and by now for at least some people in the public.

You say the card can't even be formatted / restored?
What file system did you use?

The firmware update process is a simple read operation.
The electrical design is simply reference for 1 bit operation sd_mmc mode.

ryanm101, 2021-02-03:
it could be fake sd cards, they are actually a real problem.
They can look legit, basically they are lower capacity card modified to
report a higher capacity but when you write past the real capacity it
breaks the card.

ryanm101, 2021-02-03:
Also try mounting it in a linux machine and see what dmesg outputs when it tries to mount / read it. 

if linux detects it try running `sudo dd if=/dev/zero of=/dev/<MMC DEV> bs=1024 count=10` that will overwrite the 1st 10Mb of the card and remove any filesystem info etc. 

from windows you can try 'run powershell as admin'
then
`diskpart`
`list disk`
`select disk <??>` disk number of card
`clean`
`create part pri`

remove and reinsert the card and if detected should prompt to format.

bleggett, 2021-02-04:
@ryanm101 Yeah - the SD doesn't even show up as a drive during `list disk`.

I'm OK with chalking it up to counterfeit SDs - I know it's an issue, just the coincidental nature of it weirded me out.

closed 2021-02-04 by bleggett

ramapcsx2, 2021-02-04:
I didn't think of counterfeit SD cards, but that would explain it best, yea.