x-station/xstation-issues #65: Is this a finish product?

marcopsf79, 2021-02-02:
Hi, I've installed the xstation on my PAL Model SCPH-5502, check everything, soldering point continuities, 3 different sd cards and NOTHING.
Some times it loads the firmware and detects the games, sometimes if get stuck in the Sony logo...once it loaded one game (castlevania SoTN)...only the main screen and then all black screens. Is this a joke?
This cannot be a finished product.
I will try other sd cards, and most probably other consoles, but I know what I'm going to do. I will return this crap.
I don't know what to say....The installation is quite simple...I don't get it. I don't even know how to start troubleshooting because so many things already happend.

MF

jamesrpatterson, 2021-02-02:
What SD card are you using? Have you managed to try any others?

I know it may be a dumb question but you're sure you properly inserted the SD card? It's a spring press-in type that latches, sometimes people miss that and don't push it in all the way.

birdybro, 2021-02-02:
Your unwarranted rudeness aside...

1. Can you take pictures of your installation so we can see if anything looks out of the ordinary? A continuity test doesn't mean it will perform perfectly under load, with some heat introduced, and the case reassembled, etc... Sometimes I have noticed cold solder joints, long exposed wire to the APLL, and bridges all causing intermittent behavior, which can trick people into thinking their install is fine since it works sometimes. 
2. What is the make/model of your SD card?
3. Does the estimated temperature of the system (how long it's been turned on before reset) have any impact on the behavior you are observing?

marcopsf79, 2021-02-02:
WoW, Sorry about the post. I've been truly unfortunate to install this product in faulty or dying PS1.
My frustration came from experience different issues one after another without any kind of explanation or apparent reason.
At the end the console just stopped working (don't know why).
Fortunately I have several PS1 model 5502 and I uninstalled the board from the faulty console and voilá, Everything works smoothly.
I don't know what to say, it was really a massive frustration, I should have not post this, but I found other posts with similar problems and I went nuts.
My expectations where very high for the Xstation and the product truly correspond to it. It's working flawlessly.
Maybe other people that are having a lot of issues may have just a faulty old PS1.
Again sorry about my rudeness and keep up the good work. I will keep following you guys!
Cheers,
MF

ramapcsx2, 2021-02-03:
Enjoy :)

closed 2021-02-03 by ramapcsx2