x-station/xstation-issues #66: PS1 SCPH-5502 Stuck/Freeze at Sony Logo Screen

marcopsf79, 2021-02-04:
Hi,
So I'm working with the Xstation for a couple of days and I had a problem with the first console where I install the Xstation.
Yesterday I change the ODE to another console and everything went smoothly. 
So, I was testing some, playing around, everything good and I notice some weird freezes that were happening in game. Nothing special.
Then after some time I started to get the problems I had with the first console. So now the console is stuck in the Sony logo, no sound and if freezes. I does no load the bios, nothing.
I troubleshoot everything, more than one time and everything is ok. I'm going a little bit crazy out here. Are all my PS1 consoles broken? Does it have to do with the Xstation?
So at this point I have the Xstation connected to a console that don't even loads the BIOS. I've change the power supply to another one and nothing. I don't know what to do....do you have any idea or is my problem and I'm having all my PS1 faulty?
I have another one with the CD-Rom working and all, but this is the reason I bought the Xstation, so I can keep my original PS1 and play with the Xstation on other PS1. 
Sorry guys, I've search everywhere and honestly don't know what to do.
The console was working flawlessly, perfectly for some hours, and now is dead....and it's the second one. Am I that unfortunately?

ramapcsx2, 2021-02-04:
I think this is a QSB installation problem. Probably one or two of the solder points are barely making contact.
Try to redo all the solder points using a fine enough iron tip that you can (very briefly!) dip into the QSB through hole points.
This ensures that the solder reaches to the PSX solder points and makes good contact.

Another mistake people sometimes make is forgetting to remove the resistor on the PSX CPU oscillator.

marcopsf79, 2021-02-04:
I've confirm all the continuity points....all. I've follow all the test points for the PU-18 3 times now. All seems alright.
Even if I get all the point ok, you think I should reflow everything again? I've remove the resistor.
But ok, I'll check all again...

jonasongithub, 2021-02-04:
Testing for contiunity might give you false positives because the qsb is touching the pad without being properly connected.

marcopsf79, 2021-02-04:
Ok, I will do my best probably only next Monday, but ok. I don't want to go to a new PS1. Maybe I'll remove the QSB completly and solder everything again.
Thanks for the support! I'll give some feedback.
MF

marcopsf79, 2021-02-04:
Well, I could not resist and try the installation on my 3rd and last console and everything is running smoothly for now.
Probably the other two consoles were faulty. Never have I gone through something like this. Sorry to doubt about the quality of your product.
For now I have tested some games and found some small sound glitch's in some games (Castlevania SOTN and Lunar). The music stops and a couple of seconds later starts again. Any ideia of of this can be?
I will let the console run some games for some hours to see if it fails at some point. I truly hope not!
MF

ramapcsx2, 2021-02-05:
We just investigated the extenders and they cause too much stress on the micro SD slot.
You need to be very careful with it every time you insert or remove it, or else it damages the delicate pins in the slot.
I recommend being extra careful with it from now on, to avoid any more damage.

For the music stopping, that sounds like an SD card too slow problem. It could also be a bad format.
I recommend you format the card with the official tool: https://www.sdcard.org/downloads/formatter/

If you have different cards, try a couple more and see whether that solves the problem.
The games you mentioned normally don't drop out audio. (None do :) )

marcopsf79, 2021-02-05:
I will never ever, remove the extender connected to the xstation. I will only upgrade or change the micro sd cards on the other side.
I' using some old Samsung micro SD cards but I'm waiting for some Kingston and SanDisk Ultra (the red ones) cards to arrive, so I will stick with the one that works the best.
I've read some where that the SCPH-5502 motherboard are not very reliable, so I will assume I was really unlucky and busted two motherboards. Anyway, not a problem, I can find them at 5 or 10 EUR in Portugal, just need to be alert.
I'm very happy now with the xstation and will follow closely the new developments here on forward.
Hope you can add more features, like ability to dump memory card saves to the SD card and others.
Cheers, keep up the good work and sorry about my frustrations. You all were very kind and I appreciated that!
MF

closed 2021-02-13 by ramapcsx2