x-station/xstation-issues #67: Black Screen

DarkSparkish, 2021-02-06:
So I installed XStation on my SCPH-5XXX-unit PS1, and I only get a black screen when turning on the unit with a blue light on the XStation only briefly appearing before disappearing again. This happens without the XStation connected too. The PS1 itself still lights up green regardless, but no video of any kind is showing. But when I remove the yellow wire that I used connect the QSB to the motherboard, I don't even get a black screen at all. Could the wire I'm using have anything to do with it? It looks like a pretty flimsy wire. 

Maybe soldering could be the case, but I'm optimistic that the wire I'm using doesn't provide enough power for the PS1 to boot or for the XStation to work considering I get absolutely nothing when I remove the wire. Is there any specific wire that is needed for the XStation?

retrofank, 2021-02-06:
Sounds like a poor QSB install or the pins on the DSP (bridged or others loose from the pads, etc) and you will always get a black screen, if the device is not connected, since you isolated pins from the DSP chip.  The wire is not for power, but can cause issues if it is not soldered correctly and is blocking the grounding points (Square blocks on the PCB.

I suggest you check over the pins and reflow the QSB and check for continuity.

DarkSparkish, 2021-02-06:
I see. I'll still get a new pair of wires just in case, but I'll look over the soldering too.

ramapcsx2, 2021-02-07:
Just in case you haven't seen the guide yet: https://github.com/x-station/xstation-releases/blob/main/xStation_Installation_Guide_english_PU18.pdf

Everything has to be connected and there must be no bad solder joints, or else you get things like just a black screen.

the88g, 2021-02-10:
I'm having a similar problem but no lights on the xstation at all. I've checked continuity between the flex cable pins and the matching pairs on the motherboard. I have noticed if I check ground with any of the other pins, I get a reading of 500-700 on the multimeter rather than 0 or 1 - should I expect this? Does this mean ground is bridging another pin somewhere along the line?

I will check the DSP pins tomorrow, it is possible one could have become unseated and is simply resting.

bahamutfan64, 2021-02-10:
If you’re not getting any lights on the XStation, I would check the fuses on the motherboard.

the88g, 2021-02-10:
Brill. Thank you for the tip. It was the 1a 8v going to the optical pickup. It is now booting correctly :)

ramapcsx2, 2021-02-10:
@the88g Fuses always blow for some reason. I'd check and make sure you don't have any metal nearby exposed wire.

ramapcsx2, 2021-02-13:
@DarkSparkish how did it go?

closed 2021-02-21 by ramapcsx2