x-station/xstation-issues #68: Audio stutters in some games

Jurgerna, 2021-02-06:
Hi, I wanted to make a bug report. Nothing major really. xStation is AWESOME!

I'm using USA SCPH-1001 with LATE PU-8 motherboard with RePSX replacement PSU by Rexus Nexus. (I'm using this USA console in Europe).

Everything has worked like a dream on almost everything I have been playing. Except:
- In The Hunt (USA, Redump) - major audio stutter in music in the start screen and in the game. (Arcade music)
- Captain Commando (JAP, Redump) - audio stutter especially when the map of levels is shown, sometimes while playing too.

All the best
Jurgerna

ramapcsx2, 2021-02-07:
Audio should not stutter in any game, and I know it normally doesn't in Captain Commando.
I recommend you try reformatting the sd card with the official sd association tool (if it wasn't already), or trying a different sd card.

ramapcsx2, 2021-02-10:
I've checked out Captain Commando and found one issue, which is that the tune after boss 1 randomly starts late.
I still have to check with a real CD, but I already use this game for testing for graphic corruption. It has known problems in many emulators.

In the hunt seems fine, if the low quality audio is expected.

Any updates from your side?

closed 2021-02-13 by ramapcsx2