x-station/xstation-issues #69: Weird little issue

InvaderZimm, 2021-02-09:
Ok so last night well I was playing with my xstation i was loading a game and reset it mid load and now when I try loading a game it shows loading page but does not go any further and when I hit reset It loads the game I chose but when I reset the system it keeps reloading the game but not going back to the loader like it used to and all the options for resetting from my ps1 digital do the same thing as pressing the reset on the console it reloads the game. I have tried different sd cards and reformatted the one the was in the console when this happened still the same results its like something got messed up on the xstation itself when I reset the console mid load

InvaderZimm, 2021-02-09:
And it also freezes while trying to update the firmware as well

ramapcsx2, 2021-02-09:
Hey,
this sounds a bit like your reset button is bad (on the console). 
With the console off, try pushing it several times and see if that helps.

InvaderZimm, 2021-02-10:
I can quick load games and they work but when I hit reset it goes back to the game again.
And I just hit reset a bunch and its still doing the same thing

ramapcsx2, 2021-02-10:
If the buttons (lid switch and reset button) are okay, then the fault is with the QSB installation most likely.

InvaderZimm, 2021-02-10:
It might be my lid switch because when i was a kid I glued it down so ill pull one off a dead board and install it to see if that helps

InvaderZimm, 2021-02-10:
I used to do disc swapping to play my back ups so I glued it down I was like 10 when I did this now 20 years later it might be the one thing messing this up 😆 great

closed 2021-02-21 by ramapcsx2

InvaderZimm, 2021-02-21:
I still have not found a fix for this issue why is it closed