x-station/xstation-issues #70: Unable to Boot Up Tomb Raider

RooskiWolf, 2021-02-10:
I've tried at least 50 games so far and all seem to work fine.  But when I try and boot up Tomb Raider, I get the following error:

NOT A VALID CD OR THE DRIVE IS LOCKED!
PRESS X TO EXIT

I have tried Tomb Raider versions 1.0-1.6.  Same error every time.

ramapcsx2, 2021-02-10:
Sounds like a bad rip or broken .cue sheet.

RooskiWolf, 2021-02-10:
I just tried 3 different rips from 3 different sources and still the same error.  Over 50 other games are working fine.  Here is the .cue sheet:

FILE "Tomb Raider (USA) (v1.6) (Track 01).bin" BINARY
  TRACK 01 MODE2/2352
    INDEX 01 00:00:00
FILE "Tomb Raider (USA) (v1.6) (Track 02).bin" BINARY
  TRACK 02 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 03).bin" BINARY
  TRACK 03 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 04).bin" BINARY
  TRACK 04 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 05).bin" BINARY
  TRACK 05 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 06).bin" BINARY
  TRACK 06 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 07).bin" BINARY
  TRACK 07 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 08).bin" BINARY
  TRACK 08 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 09).bin" BINARY
  TRACK 09 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 10).bin" BINARY
  TRACK 10 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 11).bin" BINARY
  TRACK 11 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 12).bin" BINARY
  TRACK 12 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 13).bin" BINARY
  TRACK 13 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 14).bin" BINARY
  TRACK 14 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 15).bin" BINARY
  TRACK 15 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 16).bin" BINARY
  TRACK 16 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 17).bin" BINARY
  TRACK 17 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 18).bin" BINARY
[... 134 lines cut ...]
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 52).bin" BINARY
  TRACK 52 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 53).bin" BINARY
  TRACK 53 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 54).bin" BINARY
  TRACK 54 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 55).bin" BINARY
  TRACK 55 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 56).bin" BINARY
  TRACK 56 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00
FILE "Tomb Raider (USA) (v1.6) (Track 57).bin" BINARY
  TRACK 57 AUDIO
    INDEX 00 00:00:00
    INDEX 01 00:02:00

ramapcsx2, 2021-02-11:
It's weird, I can't reproduce it here. I've tried with 2 different NTSC + 1 PAL version, it's always fine.
I can't think of any reason it should fail for you ><

RooskiWolf, 2021-02-11:
Turns out Long Boot works fine.  Which is strange because I thought I had tested that prior.  It appears to all be working now.

ramapcsx2, 2021-02-11:
If you find out more, feel free to amend this issue.

closed 2021-02-11 by ramapcsx2

TOPmusicman8, 2021-03-04:
Hi all,

I can vouch for Tomb Raider refusing to start a new game. I've tried a rip of my Greatest Hits version and all other versions. It will start, but once you start a new game, there is a black screen. It will load via fast or full boot but still same result when starting new game.

RooskiWolf, 2021-03-04:
I have not had a problem with any rips since I started only doing Full Boots and not Fast.

TOPmusicman8, 2021-03-04:
> I have not had a problem with any rips since I started only doing Full Boots and not Fast.

Did you use V1.6?

ramapcsx2, 2021-03-04:
The audio tracks are not working then. The rip must have an issue.
Maybe you can post your cuefile contents, and if each audio track is a separate file, a screenshot of these files names, too :)

TOPmusicman8, 2021-03-04:
Thank you =) I have a hunch it's something simple like that. I'll just continue the hunt to try to find a perfect version then. It's just funny because this is the only game that gives me any issue out of 525 lol. Thank you for developing such an awesome product and the support here!!

TOPmusicman8, 2021-03-04:
Can confirm v1.6 working via Full Boot only. You will always get the message saying "Not a valid CD or the drive is locked" but the system will restart and load fine afterwards after pressing "X". Very strange but just wanted to let anyone else know if they do intend to play this game.

ramapcsx2, 2021-03-04:
And it still works perfectly fine here :)