x-station/xstation-issues #71: Lifted pad

Sw1neF1u, 2021-02-11:
I lifted the pad the apll wire attaches to near the oscillator. I was wondering if there was an alternate connection point? Perhaps the via below it but I may have scorched that as well. Lesson learned. Any help would be appreciated.

ramapcsx2, 2021-02-11:
Everything can be repaired, of course.
Using your multimeter, find a solid spot to solder to, then put your wire end there.
If the underside of the board doesn't have anything left to solder to, use the via to bring your wire to the other side.

Sw1neF1u, 2021-02-11:
So I was able to solder to the via that is right below where the pad was. This is located right above the resistor that was removed. Blue light on the xstation when powered. Two quick green flashes when turned on but now just a black screen. I’ve checked continuity according to the pinout and all looks good. Guess I’ll have to go back and add a little more solder.

bahamutfan64, 2021-02-11:
If you could attach a photo we may be able to help guide you a bit.

Sw1neF1u, 2021-02-11:
Pretty sure I bricked it. Live and learn.

closed 2021-02-11 by Sw1neF1u

ramapcsx2, 2021-02-11:
It might be cheaper and less effort overall to send console and xStation to an experienced installer.
Good luck with it all!

16Bit-Guardian, 2022-01-28:
What does that lifted pad go to?

bahamutfan64, 2022-01-28:
It goes to Pin 4 of the CPU.

16Bit-Guardian, 2022-01-30:
I broke the main board trying to fit is on the Playstation poles where the Micro SD card goes. I was doing it wrong I know its my fault. Is there any chance I can get a replacement for it? It snapped in the center.
 I bought 2 and the 1st one I installed worked PERFECT!.  Then i snapped the 2nd one. It was gonna be a gift.
 Thanks any suggestions will be appreciated. 😭😳

16Bit-Guardian, 2022-01-30:
Disregard last message. Wrong email SORRY .!!