x-station/xstation-issues #72: Toy story 2 will not boot [Ntsc-u]

hmir992, 2021-02-11:
As the title says I try to start toy story 2 by quick or full load and the game doesn't start for me. It works on pc emulator so I'm not sure if it's a bad rip or xstation issue. I only have tried the Ntsc-u version.

ramapcsx2, 2021-02-11:
Will check it out!

ramapcsx2, 2021-02-11:
Hm, indeed, it's refusing to start up, with a debug message of "VSync: timeout" repeated 3 times.
It would be great to know what it does on a PU-18 NTSC using the disc.

hmir992, 2021-02-11:
Oh I'm running a Pal 5502 I believe I don't have a NTSC model of 5501 to check on. Wish I could be of more help but great to see that my xstation isn't broken. Must be some software issue not sure.

Thanks for looking into this!

ramapcsx2, 2021-02-11:
It's baffling. 
Definitely interested to find out what's wrong with it! :)

ramapcsx2, 2021-02-11:
The game runs easily on the PU-8 NTSC-J unit.
My dev PU-18 has a PAL mechacon in it.
It might just be another region lockout, similar to what they do in Wipeout 3!

hmir992, 2021-02-11:
So the ntsc u version is never going to be playable on pal systems then is it?

ramapcsx2, 2021-02-11:
I need to do many more tests before I can tell :)

hmir992, 2021-02-12:
no problem, thanks for looking into this.

blzla, 2021-02-12:
I tried the game on an NTSC PU-18 (SCPH-5500) and it runs fine with both boot modes. There are a couple seconds of black screen, followed by a minute of unskippable logos, but I assume that's normal.

Disney-Pixar Toy Story 2 - Buzz Lightyear to the Rescue! (USA)
SHA-1: ef20f5fcae59c80dd1d0d1bfc51ba51d8c64e967

ramapcsx2, 2021-02-12:
If it has a check similar to Wipeout 3, then it will work on machines with an NTSC-J and an NTSC-U mechacon, but not a PAL unit.
The game is made by Traveler's Tales, who are known to add special region tricks ;p

hmir992, 2021-02-12:
Thanks all , appreciate it I wonder if there is any workaround for this. Interesting Indeed

hmir992, 2021-02-12:
Out of interest you mentioned that it's a potential region lockout. Does that mean that the PAL version of toy story 2 won't work on NTSC systems? I mean does the region lockout work both ways?

ramapcsx2, 2021-02-13:
Oh, please don't come to conclusions yet. All I've seen for now is that it boots on my NTSC-J machine :)
People have analyzed the executable though, and there may be some bad code involved.

hmir992, 2021-02-13:
I'm just wondering if I boot an action replay then the game would that bypass the issue. I might try that. Thanks for looking into it.

arkfan, 2021-02-13:
Hey, I'm just dropping by to give you the solution to get the US version of this game working on PAL models. You need to apply two patchs to this game's image. The first patch is from Kalisto (KAL), the second patch is from Paradox (PDX). The second patch from Paradox is much smaller in size. Use ppf-o-matic to apply the patchs. The patchs are available on consolecopyworld.
Out of those two patchs, the one that got the game really working on PAL models is the one from Paradox which came out later and which was meant for users who had already patched their image with the Kalisto patch.

hmir992, 2021-02-13:
Nice! Thanks, will these patches keep the game at 60hz? Thanks so much for the solution I'll try it later.

JonathanDotCel, 2021-02-13:
> Hey, I'm just dropping by to give you the solution to get the US version of this game working on PAL models. You need to apply two patchs to this game's image. The first patch is from Kalisto (KAL), [...]

Thanks arkfan, that's so helpful!

arkfan, 2021-02-13:
Yes hmir992, just make sure that you choose the NTSC option on Kalisto's splash screen.

ramapcsx2, 2021-02-13:
Cheers! The Kalisto patch explains it: Game is buggy and won't work on PAL units as is.

arkfan, 2021-02-13:
You're welcome guys, glad I could help. Read the .nfo files included in the patch archives if you like. You'll get a bit of history of how Kalisto and Paradox quareled over the game not working on PAL models.
Rama : that's what Kalisto claimed, read Paradox's nfo file. Paradox later came saying no, there is mod protection on there and provided a patch to use on top of Kalisto's patch.
Toy Story 2 US doesn't like the PAL mechacon, probably the SCEE string in it. PSIO patchs that string that's why Toy Story 2 US works fine on PAL models when using PSIO. This means it's not a bug like Kalisto claimed, but an anti PAL mechacon protection.

hmir992, 2021-02-13:
Thank you! I wish there was a way around this for all games like psio has some implementations for that. A future update perhaps? Anyway, I still like xstation more, thanks guys I didn't get a chance to test that yet but I'll do it soon.

JonathanDotCel, 2021-02-13:
> You're welcome guys, glad I could help. Read the .nfo files included in the patch archives if you like. You'll get a bit of history of how Kalisto and Paradox quareled over the game not working on [...]

Hah we've just been going over that ourselves.

> Thank you! I wish there was a way around this for all games like psio has some implementations for that. A future update perhaps? Anyway, I still like xstation more, thanks guys I didn't get a chance to test that yet but I'll do it soon.

Potentially - I have a workaround to get 1.1 running  unpatched on a PAL machine, but nobody's super stoked about the idea of game-specific chunks of code if we can help it!

hmir992, 2021-02-14:
I just patched it there, yeah works great, thanks and with cheats which is nice, would be nice to get an unpatched version working on a pal machine, Thanks Guys!

ramapcsx2, 2021-02-15:
I'm really not a fan of that.
If we do it, it would be something the user has to okay before booting, maybe a dialog, asking whether to apply patches from some database.

hmir992, 2021-02-15:
Or just copy what psio does? I think the game works there. However I like your idea too yes.

ramapcsx2, 2021-02-15:
The PSIO method is not available on low level emulators.
It's probably going to be optional patches for these kind of region locks, I think :)

closed 2021-02-15 by ramapcsx2