x-station/xstation-issues #73: Pu-8 (late) and pu-18 board opinions

sak33070, 2021-02-13:
Sorry if this might be obvious to you all but i need opinions on this matter. 

I did get xstation to work but it was a challenge needless to say. 

Does anyone think pu-8 boards are just buggy or a problem case with the xstation in general? My first try was with a late pu-8 board. I'm no stranger to solder work so i thought i did a good enough job. I never got it 100% working. Sometimes it will boot into xstation. Sometimes it hangs and the console doesn't boot at all. Checked over my work twice and it didn't really change anything. 

In the end i just ended up getting another console with a pu-18 board and did the mod there. It works great now and love it. Would like to hear opinions on this because it seems REALLY strange it took a whole other console to finally get this to work. Thanks much.

ramapcsx2, 2021-02-13:
I do hear about PU-8 a bit more often, considering that most people go PU-18.
It works flawlessly whenever I install it in a PU-8 though, including my main test machine.

bahamutfan64, 2021-02-13:
Black screen upon power up is not an XStation install issue; it's an issue that can afflict **all** PS1 models. Something somewhere on the motherboard is making an intermittent connection, caused by the innumerous thermal cycles the console has experienced in its 25+ years.

It is oftentimes (but not always) fixed by reflowing the BIOS, CPU, GPU, RAM, IC305, or IC308. Basically any and/or all the chips. Whenever I've seen this issue, I reflow one chip, test, and if necessary reflow another, test, etc.

sak33070, 2021-02-13:
> Black screen upon power up is not an XStation install issue if it's intermittent; it's actually an issue that can afflict **ALL** PS1 models, modded or not. Something somewhere on the motherboard is making an intermittent connection, caused by the innumerous thermal cycles the console has experienced in its 25+ years.
> 
> It is oftentimes (but not always) fixed by reflowing the BIOS, CPU, GPU, RAM, IC305, or IC308. Basically any and/or all the chips. Whenever I've seen this issue, I reflow one chip, test, and if necessary reflow another, test, etc.
> 
> I've certainly saved several consoles from permanent/intermittent black screen issues, but there are some really stubborn consoles which will still have a black screen once every 20 boot ups that I haven't been able to fix 100% even after reflowing everything. Luckily a simple reset always allows the console to boot up normally, so perhaps not everything in life can be perfect : )

Ok oooh so it's not just me then 🤣. Good to know thanks dude. I'll see if reflowing fixes it since the QSB is still on my pu-8 board. I'm happy with the xstation on my pu-18 already but good to test and see anyway.

closed 2021-02-13 by sak33070