x-station/xstation-issues #74: SCPH-1001 install won't progress past Memory Card/CD Player Menu

WigglesEX, 2021-02-14:
Installed X-Station in SCPH-1001 Playstation with a PU-8-23 board. I've been unable to progress past this menu since finishing the install. Of worthwhile note, I did get to the "Please Insert a Playstation Disc" screen twice, but of course the third time I tried it no longer showed up and hasn't since. I get a solid blue power light when PS1 is turned on, and green access light faintly flickers twice when console is first powered on, but then does nothing after that. Access light will flash repeatedly though if I remove Micro SD. When I got to the "Please Insert Playstation Disc green" I observed the same Access light behavior as when it stalls out at the Memory Card/CD Player menu.

Here's the list of things I've tried doing to get past this menu:

-Resoldered all QSB connections
-confirmed connectivity with all solder joints on QSB
-Resoldered lifted pin pads on CXD2510Q and confirmed no contact between lifted pins and other pins.
-Resoldered all removed resister sites
-formatted microSD numerous times as both EXfat and FAT32, none of that worked
-Tried loader.bin and update.bin versions 1.04 and 1.05
-tried reset button trick that's worked for some people
-played around with microSD slot in various ways (held card in all the way at boot, inserting/reinserting during boot, stuck piece of paper in with it to increase improve connection)

The only thing I haven't attempted is to use a different microSD card (don't have one on hand) or use an extender.  

I can't help but think my Xstation's microSD slot is defective like many others seem to experience. Below are pics of my install. 

[image]
[image]
[image]

Test045, 2021-02-14:
Did you make sure the lid is closed ?
I initially lost an insane amount of time because I always test my mods before putting the top shell back, but didn't know you had to have the lid closed.

WigglesEX, 2021-02-14:
> Did you make sure the lid is closed ?
> I initially lost an insane amount of time because I always test my mods before putting the top shell back, but didn't know you had to have the lid closed.

I simulated the lid being closed by holding down the button on the MB that is normally pushed by the leg attached to the lid. Just for giggles though I reassembled the entire consoles this morning and even with the actual lid closed I still get the same behavior.

WigglesEX, 2021-02-14:
Update

I tried a different microSD card and now it's booting to the Xstation menu. Initially it was going to the "Please Insert Playstation Disc" screen, but hitting reset booted it into the Xtsation menu the next time. For the record, I was using a 64GB Scandisk microSD formatted to fat32 when it wasn't working, and am now using a 32GB Samsung EVO microSD and it works fine now.

The finicky MicroSD slot seems to be a real achilles heel on the Xstation. Castlemania, please either provide a list of known working microSD models or please replace the current reader with one that isn't so particular about cards. Thanks.

ramapcsx2, 2021-02-15:
The slot doesn't like the extenders, special care is required for those.
Using a regular micro SD, the slot can take at least a thousand cycles. 
I know from my dev units.

I think you have an installation problem yet. The reset button workaround hints at that, too.

WigglesEX, 2021-02-15:
> The slot doesn't like the extenders, special care is required for those.
> Using a regular micro SD, the slot can take at least a thousand cycles.
> I know from my dev units.
> 
> I think you have an installation problem yet. The reset button workaround hints at that, too.

Any idea on what might be causing the required reset to get to the Xstation menu? Any particular part of the install and MB modifications where I should be looking?

WigglesEX, 2021-02-15:
Update 2

Despite connectivity of soldering points being confirmed, i decided to literally resolder every single place where soldering or desoldering took place for this mod. I also thoroughly cleaned the board with 91% IPA (cleaned as I went along, but decided to get any trace of flux residue or any lingering dirt picked up). Playstation is now booting directly into the Xstation menu, no reset required now. Might have been a bad joint, maybe it was the additional cleaning, but it's good now.

However, the issue remains with the microSD cards. Samsung EVO works every time, however the Sandisk won't progress past the memory card/Music CD menu. Here is the Sandisk that won't work. It currently has the 1.05 .bin files on it and is formatted FAT32.

[image]

ramapcsx2, 2021-02-15:
In case of poor performance SD cards, you can try is the official formatting tool:
https://www.sdcard.org/downloads/formatter/

The card doesn't look like anything I know of. 
I can say with some confidence that the recent range of Sandisks all work fine.

retrofank, 2021-02-16:
That card could also, be a fake.

ramapcsx2, 2021-02-21:
So the solution here is to try a different micro SD card. Any of the recent Sandisks work for sure, nearly every other card also works, but avoid (for now) a Samsung Evo or Lexar 533 cards (only 2 lines that are known to have issues, with the Evo being partially fine).
Avoid very small and old cards. It should be fine then.
Oh, and there will be a firmware update very soon that you can install. It will further help with some slower cards.

closed 2021-02-21 by ramapcsx2