x-station/xstation-issues #75: Cannot Update Firmware

thesunnymachine, 2021-02-20:
When I try to update the firmware from 1.0.3 to 1.0.4 AND 1.0.5 I get an "! UPDATE FAILED !" Message.  

I have tried 2 different SD Cards.  1 GB and 32 GB cards.  Formatted as FAT32 and exFAT.  Using both the SD card Formatting software and the windows formatting.

I can play games just fine and I have checked all the solder points with a multimeter according to the picture on the readme and I get continuity on every point.  I purchased from Castlemainia Games.

I have a 256GB card coming that I will also try when it arrives tomorrow.  I have no issues outside of not being able to update the firmware!

Any suggestions on how I can get the firmware to update properly?

ramapcsx2, 2021-02-20:
Check your lid and reset switches, try to pushing them several times both. There is little that can cause this on an otherwise working xStation, so I'd suspect the switches.
Is this on a PU-18?

thesunnymachine, 2021-02-20:
I figured out the issue. 

I didn't realize that the update and loader files needed to go into the 00xstation folder because I had never fired up the system before.  I formatted the card and just dropped the files in the root directory, so I am guessing it was failing because there was no file to update from. 

After I moved the files into the 00xstation folder the update worked.  Didn't realize this information was in the user guide, I was reading about the SD card setup from sources that didn't include this information. 

All is well!  Thank you!

closed 2021-02-20 by thesunnymachine

ramapcsx2, 2021-02-21:
Oh, that is unfortunate. I added a warning about this when the update fails due to missing the file, but it came after 1.0.3.
Sorry about this!