x-station/xstation-issues #76: Gran Turismo 2 black screen

kingchaz1973, 2021-02-21:
Hi.
I’m trying to get the USA version of Gran Tourismo 2 tuning on a pal PS1 with xstation installed.
Game  loads up then when I select a race mode like Rally, I select track click ok then just get black screen ? 

Any idea why, is the game like Wipeout 3 USA version and won’t run on a pal system ?

birdybro, 2021-02-21:
https://github.com/x-station/xstation-issues/issues/2 - Wipeout 3 (USA) has region lockout for PAL. The game has to be patched to run on PAL systems.

https://github.com/x-station/xstation-issues/issues/72 - There are other games like this as well, like Toy Story 2 (USA).

Both are known and not likely to be something that the x-station is capable of working around, but both can be patched. ConsoleCopyWorld has the patches for games with this issue, and they should work after that.

https://consolecopyworld.com/psx/psx_patches_w.shtml

https://consolecopyworld.com/psx/psx_patches_t_3.shtml

ramapcsx2, 2021-02-21:
I'm not sure about PAL systems specifically, but I don't think they have special protections.
The game (simulation and arcade disc) works fine here.

Please make sure your setup works with other games as well.

kingchaz1973, 2021-02-21:
> I'm not sure about PAL systems specifically, but I don't think they have special protections.
> The game (simulation and arcade disc) works fine here.
> 
> Please make sure your setup works with other games as well.

Hi, thanks for it the replay. It all works great. The pal Version runs fine, I’ll look into patching the game if I can.

kingchaz1973, 2021-02-21:
> #2 - Wipeout 3 (USA) has region lockout for PAL. The game has to be patched to run on PAL systems.
>
> #72 - There are other games like this as well, like Toy Story 2 (USA).
>
> Both are known and not [...]

Thanks the reply, i will look into the patching of games.  Hope it’s not to taking lol.

ramapcsx2, 2021-02-21:
I don't think this game has region lockout code (those are pretty rare).
It is probably a different issue.
All I can tell now is that the game should run fine.

kingchaz1973, 2021-02-21:
> I don't think this game has region lockout code (those are pretty rare).
> It is probably a different issue.
> All I can tell now is that the game should run fine.

Hi, the Pal version works fine . It’s only the NTSC version that doesn’t. ?   All my other NTSC games all work fine, not I’m not sure what’s causing it. Like Iv said the load loads it’s only once a track as been selected the screen just goes blank. I am using an XRGB mini to connect it tv . Not sure that would be the problem. Tho.

ramapcsx2, 2021-02-21:
The game does change the video mode from interlaced to progressive (240p) at that point.
You would not get any display for 90% of other NTSC games either if that were the problem though.

kingchaz1973, 2021-02-21:
> The game does change the video mode from interlaced to progressive (240p) at that point.
> You would not get any display for 90% of other NTSC games either if that were the problem though.

I think Iv found the issue, it’s defo the xrgb playing up, Iv unplug the scart and plugged back in and get a piture again ?

ramapcsx2, 2021-02-21:
Ah, that's unfortunate with the XRGB. Maybe it has issues with interlace/progressive changes, made worse by the PAL system and its more off-spec timings.
Okay, at least this is unrelated to xStation. It's a scaler problem ><

kingchaz1973, 2021-02-21:
> Ah, that's unfortunate with the XRGB. Maybe it has issues with interlace/progressive changes, made worse by the PAL system and its more off-spec timings.
> Okay, at least this is unrelated to xStation. It's a scaler problem ><

Yeah, defo an XRGB sync issue. Thanks for the reply’s guys.

closed 2021-02-21 by kingchaz1973