x-station/xstation-issues #77: Green Light Blinking at Steady Pace

Faganj-y11, 2021-02-21:
I have just finished installing my kit into an SCPH-5002 PSX (PU-18 mobo) and seem to be having a similar issue to others in regards to the SD Card Slot as #44 . Blue light is solid, green light blinks at a steady pace the entire time the console is powered on. When the lid is closed or the lid button is depressed then it takes about 20 seconds before going to the memory card screen. When the lid is open or the lid button is in it's natural state, it goes straight to the memory card screen as normal. I can interact with the memory card as usual, the controller works fine in that menu and in my PS2. 

In terms of SD cards, I have tried a Sandisk Extreme (64 GB) and a Sandisk Extreme Pro (128GB) card, trying both FAT32 and exFAT on each. Below is what the root folder looks like:
[image]
And inside 00xstation:
[image]
The folders are the same both before an after insertion.

I have also tried using the extender on the SD Card mount but still the same issue. https://github.com/x-station/xstation-issues/issues/54#issuecomment-761725543 mentioned adding post it notes to the extension cable and found no difference. Might try adding a layer or two of tape and see if that gets a better result.

The slot mechanically appears to fucntion fine, locking the SD Card and exnteder into place. https://github.com/x-station/xstation-issues/issues/49#issuecomment-753126413 mentioned pushing the micro SD card in more to make contact but this hasn't worked out for myself.

Hopefully this is something completely different as I'd rather not want to worry about handling an RMA with an ocean between us. Multi-meter is arriving tomorrow to double check my joints on the QSB and to test the FFC for damage assuming there is an issue that is somehow causing the xStation to not read independent of the SD Card slot.

ramapcsx2, 2021-02-21:
At this point, I assume the extender has already destroyed the slot. We just found out what the problem with the extenders is, see this post on Twitter:
https://twitter.com/DragonsHoardMod/status/1362465300772491280

There's still a chance your slot pins or SD card pins are just dirty, but I think it's going to have to be an RMA.

Faganj-y11, 2021-02-21:
Thanks for replying Robert and sadly the pins are pristine. Do you have any details about costs in relation to RMAing with Castlemaina if any considering the nature of the situation? No worries if you don't, just mostly working out whether it would be significantly cheaper to simply try replacing the slot myself.

Guessing I'll want to file a bevel into the extension cable regardless of what means are taken to sort out the slot.

ramapcsx2, 2021-02-21:
In one of the pictures on Twitter, you can kind of make out the bent pins from the backside of the slot.
Maybe you can see those (if they are bent) on yours?

I recommend you contact Castlemania Games for procedures if you need the swap.

Faganj-y11, 2021-02-22:
I'm having a hard time seeing anything in the slot and my pictures are all turning out crap. Might see if daylight helps check but the 6th pin from the ribbon cable side appears to be bent. At the very least looking down into the slot that one is different to the others.  Submitted a ticket with Castlemania Games with a link to this issue for reference. Would rather see what they say before taking apart the slot for any proper inspection.

mattd121, 2021-02-23:
I'd be interested in knowing if you hear back from Castlemania. Been 6 weeks here and I've heard nothing from them.

Faganj-y11, 2021-02-24:
> 
> 
> I'd be interested in knowing if you hear back from Castlemania. Been 6 weeks here and I've heard nothing from them.

Weirdly enough Matt not long after you posted that message I got a reply saying to post it back to them! May want to double check spam or try making a ticket with the subject mentioning xStation RMA or something along those lines. If you've directly emailied the support email address then the lack of a ticket in their system could be why there has been no response. The ticket system is here: http://support.castlemaniagames.com/support/tickets/new

Outside of being a one time international customer I have no other affiliation to them or deep knowledge of how they work so my vague suggestions are just that. Good luck though!

closed 2021-02-24 by Faganj-y11

Faganj-y11, 2021-02-26:
> 
> 
> I'd be interested in knowing if you hear back from Castlemania. Been 6 weeks here and I've heard nothing from them.

I have had one email from them letting me know they acknowledged the issue and a replacement board has been sent off. There has been another to everyone regarding the issue as well. Was informed I had to pay for shipping and then have it credited back since I was overseas and just sent an email with the shipping recipt and tracking number. Didn't reply to an email asking about how to package the board so just used my own intuition once the replacement was sent off.

The guy I got the adapter from has also sent me the clamp for free which is pretty nice. Hopefully everything works ouy now.

mattd121, 2021-02-26:
Thanks for the update and glad to see you got taken care of. I've been completely abandoned, even Robert contacted Ryan about my situation. Still... nothing. So strange that some people get full support and other people like me get completely abandoned. I did everything by the book. Oh well...

ramapcsx2, 2021-02-26:
@mattd121 Your ticket / order number please. I don't know how you keep falling through the cracks.

mattd121, 2021-02-26:
Order #C26137. Thanks.

mattd121, 2021-02-26:
Just got the email from Castlemania that the replacement board is being shipped. Thanks @ramapcsx2.