x-station/xstation-issues #78: Weird little issue 2

InvaderZimm, 2021-02-21:
Weird little issue #69

 Closed

InvaderZimm opened this issue 13 days ago · 7 comments

 Closed

Weird little issue#69

InvaderZimm opened this issue 13 days ago · 7 comments

Comments

 InvaderZimm commented 13 days ago

Ok so last night well I was playing with my xstation i was loading a game and reset it mid load and now when I try loading a game it shows loading page but does not go any further and when I hit reset It loads the game I chose but when I reset the system it keeps reloading the game but not going back to the loader like it used to and all the options for resetting from my ps1 digital do the same thing as pressing the reset on the console it reloads the game. I have tried different sd cards and reformatted the one the was in the console when this happened still the same results its like something got messed up on the xstation itself when I reset the console mid load

 InvaderZimm commented 12 days ago

And it also freezes while trying to update the firmware as well

 ramapcsx2 commented 12 days ago

Hey,
this sounds a bit like your reset button is bad (on the console).
With the console off, try pushing it several times and see if that helps.

 InvaderZimm commented 12 days ago

I can quick load games and they work but when I hit reset it goes back to the game again.
And I just hit reset a bunch and its still doing the same thing

 ramapcsx2 commented 11 days ago

If the buttons (lid switch and reset button) are okay, then the fault is with the QSB installation most likely.

 InvaderZimm commented 11 days ago

It might be my lid switch because when i was a kid I glued it down so ill pull one off a dead board and install it to see if that helps

 InvaderZimm commented 11 days ago

I used to do disc swapping to play my back ups so I glued it down I was like 10 when I did this now 20 years later it might be the one thing messing this up 😆 great

 ramapcsx2 closed this 32 minutes ago

 InvaderZimm commented 14 minutes ago

I still have not found a fix for this issue why is it closed is this what happens you guys close it out and say the issue is fixed?

ramapcsx2, 2021-02-21:
Oh, there were no updates for 11 days, and the last mention was that you glued down your lid switch. I assumed this was not relevant anymore. Is it?

InvaderZimm, 2021-02-21:
I'm sorry im just very busy with my kids all day and I honestly forgot to switch out the part to test it. I just seen the email this morning saying it was closed 😆

InvaderZimm, 2021-02-24:
Ok so I added the wire to the alternate reset point and it fixed the issues with loading and reseting now I gotta figure out why the video keeps going on and off now and flashing green on my screen

ramapcsx2, 2021-02-24:
If you think it could be from the xStation, please shoot a short video that shows the issue :)

InvaderZimm, 2021-02-25:
I think its the ps1 digital now

InvaderZimm, 2021-03-02:
Its all fixed thanks for the help everyone

closed 2021-03-02 by InvaderZimm

ProcessReliability, 2021-05-15:
@InvaderZimm what did you do to fix this? Exact same thing started happening to me last night (2 second press no longer brings me back to the loading screen & firmware update hangs on "wait for system reboot" screen . . .)

ramapcsx2, 2021-05-15:
@ProcessReliability: xStation can't read the PSX Reset line anymore, nor can drag it down to force a reset.
Check and verify that "Pin 17" in this picture reaches the xStation and the power supply reset signal (it is one of the 5 wires of the power supply).
https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

ProcessReliability, 2021-05-15:
@ramapcsx2 UDaMan. Reflowed the solder joint connection to xStation and now it works again!