x-station/xstation-issues #79: Games in Black & White

sK00ter, 2021-02-22:
Gday I hope someone can help me out.

My Xstation is fitted to a PAL SCPH-5502 unit connected to a PAL Trinitron via component. The PS1 tested 100% fully functional before the mod.

Install seems to have gone well. I can get full colour in the menu and game launch is ok but all games are in black and white.

Ive tried multiple different backups in various file formats. Always the same.

Any and all would be very much appreciated.

Here is a vid of the behaviour.

https://youtu.be/8D47wOHIqe4

ramapcsx2, 2021-02-22:
Hey,
you're running Composite Video (the yellow plug signal), which only gives color if the region of game played matches the region of the console.
This is an old and widely known issue, but here are the quick fixes:
- get an RGB cable
- find and install a DFO mod for the PSX
- use a digital video solution (ps1digital)

sK00ter, 2021-02-22:
Turns out I need to use specifically PAL 'backups'.

sK00ter, 2021-02-22:
> Hey,
> you're running Composite Video (the yellow plug signal), which only gives color if the region of game played matches the region of the console.
> This is an old and widely known issue, but here [...]

Thx for the reply. See above comment. Many thx. The mod is rad.

sK00ter, 2021-02-22:
> Hey,
> you're running Composite Video (the yellow plug signal), which only gives color if the region of game played matches the region of the console.
> This is an old and widely known issue, but here [...]

Just to clarify for me please.... The RGB cable would give me colour but the DFO would still be 100% to play games without some sort of noticeable issue? Or will I need to def do both to even play NTSC games.

Many thx.

ramapcsx2, 2021-02-23:
An RGB cable bypasses the color encoding issue. You get a great picture and it'll always be in color.

The DFO mod can be installed in addition to that, or standalone. 
It makes it so that regular Composite Video works in PAL and NTSC mode, so you don't need an RGB cable for color.
The DFO also fixes some rare mistiming issues, mainly in some dance games.

For best quality and least effort, just get the RGB cable. It's enough to play PAL and NTSC games just fine.

sK00ter, 2021-02-23:
> An RGB cable bypasses the color encoding issue. You get a great picture and it'll always be in color.
>
> The DFO mod can be installed in addition to that, or standalone.
> It makes it so that regular [...]

Ah I see. I think the best RGB option is outputting from PSX to SCART and I don't have a SCART TV so maybe the DFO will be the better option...and a neat mod project. Thx again for your advice and wisdom mate.

closed 2021-02-23 by sK00ter