x-station/xstation-issues #80: Bent pins on SD slot? Hard to confirm...

JoeTheRetro, 2021-02-24:
So I've had my xstation for a couple months now. It was installed by ifixretro. Everything has worked great up until last night. I'm using a Laser Bear mount and since getting my console I've taken it apart a couple times. Last night after putting the SD extender back in and putting the top back on, it just gives me the white boot screen, then goes to the CD player/Mem card menu. After some Googling, I've seen that the SD extender could be bending pins in the slot. I was not aware of this until last night. I probably never would've used an extender or a mount, had I realized that was initially a problem. I get the solid blue power light on the xstation and then the mostly steady flashing green light while it just sits at the mem and cd player menu. 

I guess my question is, is there any easier way to tell if you have bent pins? It's so hard to see inside of the slot, near impossible to tell if your pins are messed up. I opened a ticket with Castlemania, but obviously, it'd be nice to know for certain whether it's my pins causing the issue or not. I'm assuming that's the problem because I obviously haven't touched any of the actual install, like the QSB or chip pins. And it just quit working now after I removed and reinstered the SD extender. Seems rather obvious that something with the slot is the culprit. I just wish I could see inside of the slot better to 100% confirm they're messed up.

JoeTheRetro, 2021-02-24:
And furthermore, regarding Castlemania, after digging a bit, it seems that there's a lot of people who say Ryan takes weeks, if not months, to respond to ticket support, if he even responds at all. There's people who claim to have submitted tickets 6-8 weeks ago and still haven't gotten replies. That's very frustrating to know. At that rate, I may as well just buy another one, since he'll apparently take your money right away, but not answer people when they need assistance.

bahamutfan64, 2021-02-25:
Even shipping XStations takes forever. I’ve ordered multiple units two weeks ago and there have been no responses to any of my inquiries regarding the order status. 

Even a simple statement on the website saying: “please note these are back-ordered and so may take longer than expected to arrive” would do wonders. Instead the XStation description reads as if units are immediately ready for shipment. 

Very unprofessional.

JoeTheRetro, 2021-02-25:
Ohhh, even more wonderful! This should be fun then! LOL!

CastleManiaRyan, 2021-02-25:
hey @JoeTheRetro if we missed a message from you we truly apologize
what is your ticket or order#?

CastleManiaRyan, 2021-02-25:
@bahamutfan64 your order should be shipped and most orders ship same or next day. What is your order#?

bahamutfan64, 2021-02-25:
Thank you for your response; order # is: C31991

CastleManiaRyan, 2021-02-25:
@bahamutfan64 got it - looks like this was put on hold in error. This will ship momentarily.

JoeTheRetro, 2021-02-25:
I closed my support ticket and just placed an order for another xstation. Order #C33390

No big deal. It's not your fault that I potentially bent the pins.

bahamutfan64, 2021-02-25:
Wonderful, thank you so much for your help! 

Anything I can do in the future to avoid having this issue, as I am a frequent repeat customer and all prior transactions have been excellent?

CastleManiaRyan, 2021-02-25:
@bahamutfan64 no just human error on our end - truly apologize

bahamutfan64, 2021-02-25:
Looking forward to receiving them! Thank You.

JoeTheRetro, 2021-02-25:
Thanks for resolving this, Ryan. You certainly didn't have to as it wasn't your fault. I certainly wasn't expecting a free replacement. I was totally fine with purchasing another one.

closed 2021-02-26 by JoeTheRetro