x-station/xstation-issues #81: Stuck at white screen and that's all

imacomputa, 2021-02-27:
Hi,
I installed my xstation last night on my PU-18 motherboard. 
No matter what I do I can't get the playstation to do anything other than boot the white playstation screen and then it hangs indefinitely.  The blue light on the xstation is on, and the green light blinks once or twice when it first power up and then goes away.
I've tested continuity of each point, I've reflowed several times, and I eventually desoldered the QSB the best I could (but was unable to remove it) and re-soldered every joint.

I'm not great at soldering, I've done some mod chips and other things here and there, but I just can't tell what I've done wrong. 

Does anyone have any tips on what I can try next?

Thank you so much.
[image]
[image]
[image]
[image]

ramapcsx2, 2021-02-27:
Hey,
The pins you mentioned are common, that's correct.
The QSB looks to be pretty far up, which makes it easy to have bad connections.
Always put a little pressure on the QSB when tacking it down, so that it's flat onto the mainboard.

Try checking for continuity between the various QSB solder points.

imacomputa, 2021-02-27:
Thanks for the response! I was afraid the qsb was not quite in position :(
I did test each point using the test point diagram posted here and they were all good. 

Maybe I have a short because of the misaligned qsb? I suppose I'll need to figure out a way to reposition it. I was just using a desoldering braid and had a tough time removing all of the solder from the middle points.

imacomputa, 2021-02-28:
Hi again, 
So I removed the QSB and resoldered it. I saw that this trace on the qsb
[image] appears to be broken and was really thinking after jumping this that it would work, but I still have the same problem.
I'm not really sure what to try at this point. 
I made sure to apply pressure on the QSB this time and it seems to be fit snugly against the motherboard.
[image]

ramapcsx2, 2021-04-06:
I forgot, did we solve this on Discord?
If not, what's the status? :)

imacomputa, 2021-04-06:
Oh yes, sorry about that. It ended up being a broken leg on the IC.
I've moved on and installed the xstation on a different psx.

Thanks for all of your help.

ramapcsx2, 2021-04-07:
Cheers, good you got it in the end :)

closed 2021-04-07 by ramapcsx2