x-station/xstation-issues #82: NTSC-U version of Toy Story 2 stuck at black screen after PS Logo

Katcher121, 2021-03-01:
I have the X-Station installed in a PAL SCPH-5552 unit. All the games, with exception to Wipeout 2097, are NTSC-U titles. Those games are:

- Castlevania Symphony of the Night
- Metal Gear Solid
- Parasite Eve
- Wipeout 1
- Nickelodeon Rugrats: Search for Reptar
- Toy Story 2: Buzz Lightyear to the Rescue

All but one of these titles work without major issues, that title being Toy Story 2. Selecting "Full Load" goes through the usual rigmaroll, the classic SCE startup then the PS1 Logo and region indicator. Unfortunately as soon as that's over all I'm greeted with is a black screen. Considering it's the only game on the SD card having this issue I can safely assume the SD card isn't the problem. Before I obtained the ODE the game in question would load from a CD-R no problem.

ramapcsx2, 2021-03-01:
This is a region lockout. We found that out here: https://github.com/x-station/xstation-issues/issues/72

The lockout is for the NTSC versions of the game when played on a PAL console.
You can find old patches for it that make it work though :)

Katcher121, 2021-03-01:
Interesting. I've never had that lockout issue when I had a Mayumi v4 in the system and burned the game to CD-R

ramapcsx2, 2021-03-01:
That would be surprising, as the game code clearly has the lockout in it.

Katcher121, 2021-03-01:
Would it be possible to look at the Mayumi v4 code or whatever to see why that makes the game run and possibly come up with a solution the X-Station can use in a firmware update?

ramapcsx2, 2021-03-01:
The Mayumi modchip simply unlocks the drive in the same way as an xStation does. There is no reason the game should work on an Mayumi unlocked drive, but not on xStation.
I assume there is another factor.

Katcher121, 2021-03-01:
Unfortunately I have no idea what it could be then. I did patch the game like mentioned in the issue you linked and it runs. I just thought you could maybe work out a solution that wouldn't require patching an games bin