x-station/xstation-issues #83: XStation hangs on white Sony bootup screen before going to MC/CD screen. All troubleshooting unsuccessful.

keugene-sg, 2021-03-02:
Am at a loss at what else I can attempt. **The XStation worked upon initial installation** but once I put everything back together it stopped working. Upon power on the power light (blue) is constant while the access light (green) only blinks twice very quickly upon initial power on & nothing else. SCPH-5500 Japanese console with PU-18 board.

1. Have followed the installation instructions exactly to the point of memorization
2. Ensured the lifted pins are isolated
3. Reflowed the remaining pins on the I/C & verified there's no short
4. Reflowed the solder joints on the QSB multiple times
4. Used the test points diagram to verify all points are correctly connected. In addition, I tested using the mobo test points & the **individual QSB FFC pins**. Everything checks out fine.
5. Verified the XStation board's card slot works fine. No bent pins. Plus it worked fine during initial installation.
6. SD card is formatted correctly with the necessary "00xstation" folder & relevant files. The initial installation worked fine & games could run from the SD card with no issues so I'm 100% certain there's nothing wrong with the SD card or it's format.
7. Tried another SD card anyway from scratch. Same result.
7. Tested continuity on the FFC. No issue there
8. Yes I've read the entire GitHub issue logs (both open & closed) & tried everything suggested (less buying another console). I don't live in the US & where I'm from every clown believes anything old that's gaming related is "rare & valuable".

Am seriously at a loss. What annoys me the most is that the damn thing actually worked fine upon first installation & simply stopped working after screwing the thing back together. Any further suggestions is very much appreciated.

Thank you very much for any help that's available!

ramapcsx2, 2021-03-03:
My first guess goes towards the 2 mechanical switches that are tapped: Lid and Reset.
Can you make it work again with an open shell and something pressing down the lid switch?

Another thing is the little plastic tab that you have to cut from the case bottom. If you skip that step, the flat flex cable can get swished and damaged.

keugene-sg, 2021-03-03:
Hi Rama, thanks for the support.

Yep I've depressed the lid switch both with lid closed & by manually pushing it down with a sharp implement. Makes no difference. When the lid switch isn't depressed it boots into the MC/CD screen almost immediately after the white Sony screen. With the switch depressed it hangs at the white Sony screen for about 10s before dumping me to the MC/CD screen.

The plastic tab was also cut & filed down flat. You couldn't even tell there originally was a tab there. I also checked the continuity of the FFC everything's fine.

ramapcsx2, 2021-03-03:
Check out this issue: https://github.com/x-station/xstation-issues/issues/39#issuecomment-747551682

keugene-sg, 2021-03-07:
Hmm...correct me if I'm wrong but if any of the vias or connection lines on the mobo was evaporated/cut wouldn't there be no continuity between the mobo itself & QSB pin? I've traced all the lines to determine where each mobo test point is connected to on the mobo (mainly IC701 & IC305) & there's no issue with continuity between the I/C & QSB.

I'm going crazy trying to debug this.

ramapcsx2, 2021-03-07:
This particular via was about signal "20" here btw: https://camo.githubusercontent.com/258635023ed2e357d6bd0392ef9fe710d80258d8775c064049f82889b66053fd/68747470733a2f2f6d6d6d6f6e6b65792e636f2e756b2f77702d636f6e74656e742f75706c6f6164732f323032302f31322f7853746174696f6e2d353530582d5153422d6d61702e706e67

It should connect to one pin on the square IC (the mechacon) to the left of the QSB.

ramapcsx2, 2021-03-08:
And please check that you have a connection for the points "9" and "6" with these 2: https://i.imgur.com/rdkF6iV.png

This was something we found and fixed on a board yesterday.

keugene-sg, 2021-03-09:
Have checked connectivity for Points 9 & 6 per the diagram. Everything checks out.

ramapcsx2, 2021-05-09:
Any update?
If nothing you tried fixed it, serious debugging with an oscilloscope is required.

ramapcsx2, 2021-05-10:
Great it's working now :)
I have no control over the QSB process. I can only wait and hope the hardware design guy can find the time for it..

closed 2021-05-20 by ramapcsx2