x-station/xstation-issues #84: EDC/ECC correction feature questions

littlebalup, 2021-03-02:
Hi,
I didn't tested the 1.1.0 introducing the new EDC/ECC feature yet. That's a good idea but I have few questions.
I'm using Xstation to play games but also to test my backups images ensuring everything is fine. Including EDC/ECC. Then if everything is fine with xstation I know I can burn my backups tu use them on my other modchiped ps1.
So, can this feature be toggled in the settings? If not, can you add it to your to do list for the next release? Or maybe just a message when launching the game saying bad EDC/ECC have been patched.

Thanks

ramapcsx2, 2021-03-03:
Hm, I'm a huge proponent of not exposing options with little to no impact .
This option would be irrelevant to most users and it doesn't serve your purpose well either.
When ripping a CD, the EDC/ECC of sectors protected with it ensures that the data is good.
If there was a problem in the rip, the drive or ripping software would use these bytes to fix the bad data.
In the end, the EDC/ECC matches the data, and is thus the same as on the disc.
This is how burning discs without correction bytes produces a correct EDC/ECC set again.

Unprotected sectors and CDDA are what you need to worry more about.

To be clear: Only patched disc sectors where the EDC/ECC has not been properly restored will see any change on the fly on xStation.
Everything else, such as any regular image file you quickly create with ImgBurn, will have 100% correct EDC/ECC.

ramapcsx2, 2021-03-08:
Did I change your mind successfully? :)

littlebalup, 2021-03-08:
Yes. I agree about about regular clean rip. And, indeed, I was thinking more of patched rip cases (mostly fan traductions). I should have been more clear in my original post.

closed 2021-05-09 by ramapcsx2