x-station/xstation-issues #85: After a week of use it stopped working after a 2 second reset button press.

jvivallos, 2021-03-03:
I got the xstation working, I played a couple of games for about a week. Then when I tried to get back to the menu the xstation by pressing the reset button for 2 seconds it stopped working and now I always get the no disk screen.

Things I've done:

- Reflowed all the solder points
- Made sure the chip pins weren't shorting or still connected. 
- Tested continuity on all the QSB and the console and also between the QSB and the xStation
- Reformatted the SD card using the suggested software.
- Keep the lid open and close it once it boots.

My PS is a Japanese 5500 PU-18
I get a blue light and also some flashes of green light.

I'd appreciate any help.

ramapcsx2, 2021-03-03:
A connection became loose, that is obvious.
You need to debug traces from the flat flex connector on the QSB to their end points as best as you can.
One of them (the reset line?) failed. (pink in the attached image)
[image]

jvivallos, 2021-03-05:
Yes I have tested all of them many times. And I forgot to add on my first post that I have a cable directly attached from the QSB to the reset test point as suggested in xstation-issues.

And now I've tested continuity on each of the flex "cables" against the board test points.

[image]
[image]
[image]

tjw2005, 2021-03-13:
Have you resolved this issue? I just experienced this myself on a US PU-18 with PS1 Digital. I was playing Ace Combat 2, the music stopped playing. When the level was complete the system froze. When I restarted, the console would boot into the CD Player/Memory Card screen only. I tested all connections to the flex cable connector on the QSB using the above test point diagram. I tried reverting to older firmware. I haven’t been able to solve it yet.

jvivallos, 2021-03-14:
No @tjw2005  I haven't been able to resolve it. I'm hoping @ramapcsx2 can tell some more things I could debug. Is there a way to debug the xstation?

ramapcsx2, 2021-03-14:
I can't help much, solving install issues from afar.
If you can, send everything to an installer and have them check everything, is my best shot.

tjw2005, 2021-03-14:
> No @tjw2005 I haven't been able to resolve it. I'm hoping @ramapcsx2 can tell some more things I could debug. Is there a way to debug the xstation?

After spending quite a while trying to debug everything I circled back to the lid sensor. When I tested it with a multimeter it appeared to work, but apparently it was only working intermittently. Pressing down on it hard with the shielding off got the xStation to boot. I’m confident that this 25 year old mechanical switch is the issue on my system!

ramapcsx2, 2021-04-07:
@jvivallos Can you send it all to an installer nearby? Someone that's setup to debug the issue will be able to fix it.

jvivallos, 2021-04-09:
Unfortunately I don't have anywhere to send it here in Chile. When you mean setup to debug, do you mean oscillators and programmers??

ramapcsx2, 2021-05-09:
Sorry for the late reply, but in case you can't find a problem, all that's left to do is to send the console to an expert.
This is what I was asking: Whether you can send the console to an experienced installer, so he can debug it.