x-station/xstation-issues #86: Lid switch on motherboard help?

YodasFootPowder, 2021-03-04:
So my Xstation has been working great, however, I've found that my lid switch is a bit finicky which can lead to issues booting, obviously. I've applied a bit of Deoxit to the switch on the motherboard and that has seemed to help, however, that's clearly not going to likely be a permanent fix. Is there any way to make the console just think the lid is always closed? Perhaps by bridging 1 or 2 of the four metal contact points on the bottom of the motherboard, under the lid switch? I've done some Googling, but mostly come up empty. Or, is it possible the Xstation will be able to disable the lid switch functionality in a future firmware update? I don't even know if that's possible. Just a curious, so I thought I'd ask. It seems the lid switch is definitely an issue for many people with this hardware and it'll likely only get worse in the future as these original model Playstations get older. There's got to be some points on the motherboard or the bottom of that lid switch that can be bridged to disable the switch. I, however, am just a lowly peasant without the knowledge.

darcagn, 2021-03-04:
Yes, I did this on mine while doing XStation troubleshooting.

If you have a PU-18 motherboard, solder a wire from any ground (e.g. that big open ground plane next to it) to this point to close the lid permanently.
Wire it to a switch to switch it on and off as desired
[image]

YodasFootPowder, 2021-03-04:
^^ Awesome! So that's to point D118? Did you use a switch for yours or did you just wires yours to be always closed?

YodasFootPowder, 2021-03-04:
Alright, I went ahead and added a wire and all is well. I appreciate this info. Nice to not have to worry about a janky switch anymore!

closed 2021-03-04 by YodasFootPowder