x-station/xstation-issues #87: Xstation won’t boot at first try

novamax89, 2021-03-05:
Hello
I installed yesterday xstation to a pu18 motherboard. 
it didn’t work at first try where you can heard the booting Sony logo sound but it went to mc bios. 
At second try there was the logo with no sound and again to mc bios. 
At third try it worked flawlessly and boot to xstation. I just tried ff8 and it worked perfectly with no skips both intros. 
After the third try it worked always well after shooting down or resetting the console via in game reset mod. 
today I start the console and it booted to xstation just at third try like yesterday. 
it’s not a big deal if I have to start the console 3 times to make it work flawlessly but it is not ideal, and I fear it may cause problems in some games, I haven’t had so much time to test it well. 
I really don’t understand. Could somebody help?

ramapcsx2, 2021-03-05:
It sounds like an installation problem or an SD card problem.
Please try a different SD card and please check that everything is installed according to the guide:
https://github.com/x-station/xstation-releases/blob/main/xStation_Installation_Guide_english_PU18.pdf

If you can't find the problem, I recommend you send the console to an expert in your area.

novamax89, 2021-03-05:
It is done exactly according to the guide. I don’t understand why it works well even hours after it’s been turned off. It is my second mod chip ever but I think I’ve done things good, I checked many times. It’s not a big deal but I don’t think it is an install problem though might be wrong and not even notice

novamax89, 2021-03-05:
It’s not big deal now as it’s always working fine. I don’t think it’s worth spending money to send to an expert imho

ramapcsx2, 2021-03-06:
Well, all I can say from here is that it shouldn't behave like that.
If the xStation makes it into a game or the loader even once, then it is fine and should always work as expected.
This is why I recommend checking the only other external factors: SD card and installation.

novamax89, 2021-03-06:
I tried a different sd card and it’s behaving exactly the same. So I don’t think it is an sd card problem. I still can’t figure out,, why if it’s always working there should be an install problem, if there was a problem should never work at all. It is like the console need to warm up a bit. And if I shut it down it works again even 5 hours later. In the same install I installed a DFO, could be it causing the problem?

ramapcsx2, 2021-03-06:
Hm, I don't think a DFO could cause it, but it sounds like you've heavily modified the console.
Any mod that connects to a signal that the xStation also uses might be the cause.

novamax89, 2021-03-06:
> Hm, I don't think a DFO could cause it, but it sounds like you've heavily modified the console.
> Any mod that connects to a signal that the xStation also uses might be the cause.

Heavily modified?? Dfo and igr are very simple mods, and they should be all compatible with xstation

novamax89, 2021-03-06:
Anyway the problem is just xstation and maybe dfo. It was having this problem before I installed igr. Anyway a modder told me it could be the ps1 psu to create the problem as it works fine after condensators are warm

novamax89, 2021-03-06:
I’m switching psu. I’ll keep you updated

novamax89, 2021-03-08:
I switched the psu but after almost 24 hours the console was turned off and today I’m having the exact same problem. The console starts just at third boot and works flawlessly from then

ramapcsx2, 2021-03-08:
"At second try there was the logo with no sound and again to mc bios."
Do I understand this correctly as in, the console did not produce sound this round?

studmuffinhak, 2021-03-09:
I've been having this issue as well.  Mine usually starts after the second boot.

Ive done over 20 Xstation installs PU-18 and 2 of my xstations from the latest batch(Early Feb) I bought have this issue.  I still  cant pin it down.  I was able to alleviate it on one by directly connecting pin 17.

ramapcsx2, 2021-03-09:
What pin 17, and where did you connect it to? Was that a complete fix?

studmuffinhak, 2021-03-09:
I just hardwired it to the via on the mobo.   I wasn't able to thoroughly test both consoles but one was mostly fixed.

https://camo.githubusercontent.com/258635023ed2e357d6bd0392ef9fe710d80258d8775c064049f82889b66053fd/68747470733a2f2f6d6d6d6f6e6b65792e636f2e756b2f77702d636f6e74656e742f75706c6f6164732f323032302f31322f7853746174696f6e2d353530582d5153422d6d61702e706e67

ramapcsx2, 2021-03-09:
Okay, this "17" is the reset line. This suggests the affected consoles may have an issue with the power supply.
If you can, please try swapping the PSU and see whether it works any different.

studmuffinhak, 2021-03-09:
I tried 3 different psus from my testing and issue was still there.

ramapcsx2, 2021-03-09:
Can you try using a different xStation, see how that behaves?

studmuffinhak, 2021-03-09:
I did that as well and same issue.  They were from same batch.

<comment@noreply.github.com>
Sent: 3/9/2021 2:29:48 PM
try (#87)

ramapcsx2, 2021-03-09:
That's good. I would be surprised if the xStation was at fault, as the design is simple and can be well controlled in production.

The problem is probably with the mainboard. 
Something might be different on some boards. Something I haven't encountered yet.

bahamutfan64, 2021-03-10:
If the console doesn’t boot up normally the first time power is turned on, but then starts up upon a reset, that sounds like an iffy connection on one of the chips, maybe the CPU or RAM. A reflow may help. 

If there’s no audio, conversely that would be the sound chips IC308 and IC310.

msimplay, 2021-03-17:
Also having the same problem with a DFO + Xstation installation.
The person who modded it also left a previous PSIO board installation in there.

I would be interested in knowing what the fix for this is if it has been fixed so I can send it back for repair and show him this thread

kingchaz1973, 2021-03-21:
Iv started getting this after I installed  a DFO, first time boot it plays up then 2nd or 3rd boot it works perfect.  I have idea why.  Maybe the different timings on the DFO are casing slight issues .  Either way I can live with it as now all my cars run perfect and colours are correct. I have a pal system

msimplay, 2021-03-21:
> Iv started getting this after I installed a DFO, first time boot it plays up then 2nd or 3rd boot it works perfect. I have idea why. Maybe the different timings on the DFO are casing slight issues . [...]

Colours are correct in both PAL and NTSC modes?

kingchaz1973, 2021-03-21:
> Colours are correct in both PAL and NTSC modes?

Yes all my games bar one are ntsc.  The only pal one I have is wipeout 3 which runs fine ..

why should the colour be off ?

msimplay, 2021-03-22:
> Yes all my games bar one are ntsc. The only pal one I have is wipeout 3 which runs fine ..
>
> why should the colour be off ?

It seems my TV is having issue with DFO the colours are incorrect in PAL mode funny because the console is PAL

novamax89, 2021-03-22:
This problem is probably due to the dfo installation. I’m waiting for a new qsb from castlemania to make a new install without dfo and will post results here

msimplay, 2021-03-22:
> 
> 
> This problem is probably due to the dfo installation. I’m waiting for a new qsb from castlemania to make a new install without dfo and will post results here

The DFO was installed wrong or the settings are incorrect of the DFO itself?

kingchaz1973, 2021-03-22:
I’m getting crackle sound on first boot and little white dots on screen and now some games just hang on xstation before they load up.  Is this an install error of the DFO or conflict with Xstaion or even annxstaion fault ?

ramapcsx2, 2021-03-22:
That does sound like a bad oscillator, so yes, check the DFO.
You may also want to update the xStation, as it fixes a hanging bug. It may not be related, but this makes sure.

kingchaz1973, 2021-03-22:
> That does sound like a bad oscillator, so yes, check the DFO.
> You may also want to update the xStation, as it fixes a hanging bug. It may not be related, but this makes sure.

Hi, I fixed my issue. I took off my clock wire cleaned up the area and re fitted. I’ll report back if it goes bad again.  But so far no cracking no dots. Just smooth operation and everything is working as it should now.  
All games load first time.. so hoping it was just a cold solder or left over flux that was causing the issue.

kingchaz1973, 2021-03-23:
[image]

kingchaz1973, 2021-03-23:
This is my new install and all works perfect

ramapcsx2, 2021-05-09:
Perfect :)

closed 2021-05-09 by ramapcsx2