x-station/xstation-issues #89: Akuji the Heartless - Stuttering FMV

PsychoFox11, 2021-03-06:
This is another developed by Crystal Dynamics, who made Gex: Enter the Gecko, and I'm experiencing similar issues to what it was having. I also have a PSIO and the video plays smoothly from that.
I was first using a Samsung card and it would stutter on a frame pretty often (every 5-10 secs). Switched to a Sandisk Extreme and it isn't nearly as bad, but still happens.

Not sure if the fix for Gex was game-specific or universal. If it was game-specific, maybe this is doing something similar, and maybe this is the best it can get if it was universal? Of course it could be completely unrelated and not handling video the same, just noticed same dev and similar playback issue.

This is the opening story movie after choosing 'New Game' - I don't have any capture devices but am pretty sensitive to stuttering video etc. and there's none from PSIO using an old brandless card.

I'm using a late PU-8 board with a Laserbear mount, NTSC system and game. Happy to provide any other info (that doesn't require a capture device).

YodasFootPowder, 2021-03-06:
I just tested Akuji on my PU-18 motherboard with a Sandisk card. I noticed some very, very slight micro stutters during the opening FMV. Mine were almost unnoticeable, but definitely there. However, if a person was only half paying attention to mine, you'd probably miss it. It definitely wasn't playing properly, though.

ramapcsx2, 2021-03-06:
It's the same engine then. As we found out, the games run randomly choppy from disks as well.

YodasFootPowder, 2021-03-06:
^^^ That's a good point. There's more than a few games I've noticed FMV not running all that smooth from the disc.

PsychoFox11, 2021-03-07:
I think in many cases, like mine, after getting the XStation, a PSIO is our only point of comparison as we can't use discs anymore. I'd test from a disc if I could!
So does this need to be enabled per title, or does this game already use the updated code that helped Gex a little, meaning it's running it's best already?

ramapcsx2, 2021-03-07:
The new firmware runs the new code that places all the important events so that my test games all like it, and the Gex engine stutters the least on my test PU-18 (and -8 as well).
This is the best you'll get, and it's the same for all games (I don't like per-game settings).
The Gex engine is just flawed like that, and randomly too (simply re-run the FMV and it will perform differently).
There is nothing I can or should do anymore. It's been tweaked to the best of my ability, and this gives the best reproduction of real intended behaviour.

Gardosen, 2021-03-07:
> They struggle with compatibility, but they can score on Gex :)

No offense, but in terms of compatibility from a customer perspective, PSIO, Mode and X-Station are even if you do not look as the second value after the ",". I do not think this argument is valid.
the only difference is which games have issues. those are different for the systems because of their difference in handling games.

ramapcsx2, 2021-03-08:
No offence taken, but the argument *is* valid.

In the case of the PSIO, it can not be fully compatible with the library, due to how it has to work over the expansion port bus.
One example is that it somehow has to transfer an audio stream (XA, CDDA), but it isn't connected to the bus that should carry this stream.
However it manages to play audio anyway, it must be using resources that aren't meant to be used.
Something has got to give on that one.

The MODE is taking the same approach as the xStation, so it has the same performance in Gex.
We've checked and confirmed this quickly. The judder was visible immediately.

So the thing is, given these circumstances, the ODE should first and foremost strive for compatibility.
In the case of the Gex engine, this unfortunately means to accept the judder that is build into the engine.
I never found a variation of signal timing that fully avoids it, but I made it as good as can be, without breaking other games.

What the customer definitely doesn't want to see is a game that stops working, right?

closed 2021-03-22 by ramapcsx2