x-station/xstation-issues #91: Final fantasy viii pal not booting

Rapid117, 2021-03-09:
Hi Final fantasy viii pal version is booting to blacks screen. But final fantasy viii ntsc US is booting fine. Anyway to get the pal version working?

ramapcsx2, 2021-03-09:
Have you updated the firmware to the latest version?
The game requires support for LibCrypt, which has been added around 1.0.5.

Rapid117, 2021-03-09:
Cheers for the response. Yeah I've updated to the latest version. It's the only game I've had with this problem

ramapcsx2, 2021-03-09:
That's odd then, I'm using FF8 in my quick LC test and it always worked.

Rapid117, 2021-03-09:
I'm going to look for another pal rom of it to rule that out. Might be a bad dump

carmax91, 2021-03-14:
I can confirm too that you need a patch or subchannels data for your dump, PAL FF VIII are LibCrypt protected.

ramapcsx2, 2021-03-14:
Which PAL release is this by the way? The various languages all require a different LC patch, so maybe yours is missing somehow.
I'm testing with the PAL-G version of the game, and it works fine out of the box.

Rapid117, 2021-03-14:
I'm use'ing the english pal release from the redumps on archive.org and I've dumped my own pal black box copy aswell with no luck. Every other game I've tested not a problem just this one

carmax91, 2021-03-14:
All PAL version have LibCrypt protection. You need subchannels.
To dump with subchannels you need a reader wich supports RAW-DAO reading and CloneCD with appropriate profile, otherwise you will get a bad dump. 
As alternative you can apply a patch (i don't know if i can link the site to download it) to remove the protection.
(Maybe you can also use the appropriate .sbi files from redump.org, but i don't know if xstation support it...)

ramapcsx2, 2021-03-14:
@carmax91 The xStation has the required info included. No extra user intervention should be required. This is apparently failing here :p

carmax91, 2021-03-14:
So, you don't need the subchannel file to work?

Rapid117, 2021-03-14:
carmax91 Is all I need to do is dump my copy of FF8 using CloneCD for it to work?

carmax91, 2021-03-14:
I've tested the Italian version of the game, and i can confirm that works without any patch or subchannels data. Impressive @ramapcsx2 !  (But it's time to write a better and more complete manual mate!)
@Rapid117 at this point there is a failure with xstation, but you can play with the patched version of the game (i don't know if xstation support CloneCD .sub data for subchannels data)

Rapid117, 2021-03-14:
Ok thanks. I've patched the rom and its booting up fine. Thank for the help. I'll close this thread

closed 2021-03-14 by Rapid117

ramapcsx2, 2021-03-15:
It's an xStation problem. The PAL-E variant is missing. I'll fix it for a later release :)