x-station/xstation-issues #93: Soul Reaver strange intermittent hang

AndreasV81, 2021-03-12:
Soul Reaver NTSC at Xstation Firmware 1.1.0 hangs intermittently. Looks like a game or frame freeze. Strange thing though, that the music continues playing and that the issue is always resolved by opening the lid. The game continues instantly when the lid is opened, NOT closed. This happens sometimes before cutscenes but now in the drowned abbey all the time. This renders the game unplayable.

Easily reproducible: Teleport to the drowned abbey, switch to physical realm, run down the corridor, open the door and right before the door closes behind you the game freezes. Opening the ps1 lid then continues the game.

This might look like some trivial loading error but actually if you play the game in an already loaded room with the lid opened the game does not halt, it simply does not load new objects. 

Tested 3 different sdcards, all formatted with the format tool recommended, all the same. Game hangs at the very same position. When i revisit other areas like "necropolis" or "nupraptors retreat" this issue does not happen at all.

Tested Xenogears and FF8, both NTSC, the intro FMVs play flawlessly, so i dont have any sdcard access problems when it gets critical, like during FMVs.

PU18 board in a 5552 pal console with ps1digital also installed. The software lid switch trigger implemented in ps1digital is not installed. Neither is psnee or any other modchip installed. I also desoldered, opened and deoxidized the lid switch internally as an intermittent contact was my first guess. Things did not change.

Can anyone test this please :)

ramapcsx2, 2021-03-12:
Yeah, sure I'll test it.
I've no idea what teleporting to abbeys or realm switching means. I need help reproducing :)

AndreasV81, 2021-03-12:
Update: this also happens on a different ps1 with modchip, playing the game from CD, so it is not specifically an xstation issue ... but maybe it can still be resolved by it. By the way, the rip is good, md5sum matches the one at redump.

ramapcsx2, 2021-03-12:
Uhum, thanks for testing that!
Is there anything on the net about it maybe?

AndreasV81, 2021-03-12:
Thanks, still, maybe you can solve this issue by software. Start Soul Reaver NTSC on a pal ps1, you can use gameshark codes to open up all teleport locations, exit the first room you start in, step on the teleporter in front of the gate, press square, choose the symbol which matches the drowned abbey ( web, cannot post it here) and step through the gate. in the back of the room you see a blue fountain like thing, step on it, press select, press square, now you switched to the physical realm, because you always start in the greenish spectral realm. now exit the gate room, run down the corridor and open the door by stepping in front of it and pressing square. a short cutscene shows raziel opening the door and stepping through. the moment the door would shut automatically behind him the game freezes.

ramapcsx2, 2021-03-12:
I tried my best with cheats like "all glyphs" and whatnot, but the game has too many items and lore i have no clue about.
I need a save game :p

AndreasV81, 2021-03-12:
I will copy my save game via ps2 to usb and send it to you. How do i do that? Where should i send it? Or can i attach it here somehow?

ramapcsx2, 2021-03-13:
Will these work? https://gamefaqs.gamespot.com/ps/197763-legacy-of-kain-soul-reaver/saves

I'll load one of them up, move to that gate with the 2 platforms to stand on and select the symbol that I found on Google:
https://static.wikia.nocookie.net/legacyofkain/images/9/93/Abbey.png/revision/latest?cb=20130923125332

Then walk through the gate and do the rest. Is that right?

AndreasV81, 2021-03-13:
Exactly. Thanks. Sorry for the late response.

ramapcsx2, 2021-03-13:
Alright, I've got it to hang.
I found some mentions of this in old forums, but so far no good clue.
I'm on it! :)

AndreasV81, 2021-03-13:
Much appreciated 😊. The strange thing is that you can resume the game by opening the lid.

ramapcsx2, 2021-03-13:
It's a very interesting title for sure. The last prototype version with enabled debug menu of this (Jun_28,_1999) doesn't hang.
I'm having trouble to get to that area in even earlier versions, which also have a "CD Debug" option.
Maybe I can find out more.
It would be great if you could check your CD on other variations of the PSX!

AndreasV81, 2021-03-13:
I checked it with a PU18 board at a 5552 model with xstation and with a PU22 board at a 7502 model from cd. i could also check from cd at a PU22 in a 5502 model. and also the latter two with PSIO.

AndreasV81, 2021-03-13:
Its an absolute masterpiece, and i will do my best to coop. to solve this ...

ramapcsx2, 2021-03-13:
Did you get those same hangs on the other consoles?

ramapcsx2, 2021-03-13:
I've tested on a PU-8 NTSC-J with xStation in the mean time, and it doesn't hang.
It might be a problem with the PAL Mechacon.

AndreasV81, 2021-03-13:
I get this hang on all 3 ps1 i have from cd. But from the psio it works. It doesnt hang with the psio. If it was the mechacon it should hang there too i guess. I guess it works from psio because they handle lid open or close events by ejecting and inserting the sd card. With psio it works with a pu18 and pu22 board. It should be solvable then  i guess?

ramapcsx2, 2021-03-13:
PSIO doesn't use the original Mechacon, it behaves differently than the real machines.

I just wonder how such a big name game can hang on many different consoles, yet it's rarely talked about.. ( I can only find one forum entry for it, which advises the CD lid trick to bypass it. )

AndreasV81, 2021-03-13:
If the localization of a game was done properly, maybe there were not too many pal users wanting the ntsc version. Like in resident evil jill running in slow motion in the pal version compared to ntsc

AndreasV81, 2021-03-13:
Still ... do you think you can create a non mechanical workaround for this? Like looking into the data stream, capturing the hang situation and trigger a lid switch if the user wants? At least the ps1digital menu still pops up if triggered during the hang, so maybe i should install the ps1digital lid switch ?

ramapcsx2, 2021-03-13:
Playing into this is that no one was meant to run out of region games on their console. It is only possible with modchips or swap tricks, etc.
This may be how it wasn't discovered (as well as occurring late in the game).

I've asked users on Discord to try this on NTSC consoles. We should hopefully know more soon.

AndreasV81, 2021-03-13:
Thinking about the emulation of the data stream ... when you open the lid, the laser goes back to its starting position. So the stream is interrupted as long as the lid is open. Is this event maybe triggering a track seek reset or something? Or some other type of mechacon reset? So would it be possible to capture the hang situation, when no data arrives but should and reset seeking? Just like a read retry in case of scratched discs? I just think that even if this issue is coupled with the mechacon, in all cases i heard of, the game freezes entirely as well as the console itself. In this case however its just an intermittent hang.

ramapcsx2, 2021-03-14:
No no, it's all much simpler: The game spinwaits on a variable, triggered normally by the CD system, that never happens.
The game also aborts the spinwait when a CD error occurs.
Opening the tray is "error enough" so that it breaks the spinwait. That's all :p

I've got to figure out what the incompatibility is first, then I can think about fixes :)

carmax91, 2021-03-14:
Maybe a sorta of GPU timing error due to different PAL vs Ntsc timings?
I'm waiting for an NTSC oscillator to mod my PAL PU-18 console to have correct timings on NTSC titles, when it arrives i can do some testings if you need help...

AndreasV81, 2021-03-14:
That woul be great 😊. Can you also test the issue on an unmodded one wether you can reproduce it too?

carmax91, 2021-03-14:
When i receive the crystal I will do some test

ramapcsx2, 2021-03-14:
We've already checked and ruled out GPU timings :)

ramapcsx2, 2021-03-14:
Okay, I could confirm that Soul Reaver is working on an NTSC-J PU-18.
It is an unintended Mechacon incompatibility for the -U version of the game on PAL machine.
I don't know how to deal with this yet, but maybe a patch for the game exists that fixes it.

carmax91, 2021-03-14:
> 
> 
> We've already checked and ruled out GPU timings :)

Perfect, what a strange bug...

birdybro, 2021-03-15:
Maybe, but not likely, the PAL/NTSC selection patch for the game might work? 

from: https://consolecopyworld.com/psx/psx_patches_s_2.shtml (they sometimes have NSFW ads so if you want to avoid it I attached the zip of the patcher for NTSC Soul Reaver: Legacy of Kain.

[kal-sr2t.zip](https://github.com/x-station/xstation-issues/files/6138486/kal-sr2t.zip)

@AndreasV81 Test it and see if it works for now. Read the .NFO file that comes with it for instructions.

ramapcsx2, 2021-03-15:
I've attempted that patch, but it completely destroys my image of the game. EDC/ECC is broken, and fixing it doesn't help: The image won't start.

birdybro, 2021-03-15:
Good to know for sure.

Seems like it's super old, probably won't run on anything but Windows XP :P

ramapcsx2, 2021-03-22:
So to summarize:
The game has an (intentional or not) region dependency.
Only an NTSC console can play the NTSC game.
For cross region compatibility, the game requires a patch to be developed.

mathieulh, 2021-03-28:
[Legacy of Kain - Soul Reaver (Europe) PAL to NTSC.zip](https://github.com/x-station/xstation-issues/files/6217576/Legacy.of.Kain.-.Soul.Reaver.Europe.PAL.to.NTSC.zip)

I made an xdelta patch to convert the PAL version of Legacy of Kain Soul Reaver into an NTSC 60 version that would run on PAL consoles.

mathieulh, 2021-03-28:
So, some more news for you, I have a PAL console with PAL mechacon and a NTSC US BIOS (I did a BIOS Swap) and I attempted to reproduce the issue, unsuccessfully, the game doesn't crash at all! So I very much assume this is a BIOS issue rather than mechacon related.

AndreasV81, 2021-03-28:
Well, this sounds interesting. Where did you get your ntsc bios from? can you simply change the Chips or is there more to it?

AndreasV81, 2021-03-28:
On the other hand, if it runs on the psio ... how can it be bios related? The Psio uses the original bios as far as i know... But anyway, maybe we can track this down someday.

AndreasV81, 2021-03-28:
Which bios version did you swap to anyway?

mathieulh, 2021-03-28:
I used the US PU-18 BIOS from here:

https://willsconsolemodifications.co.uk/ps1_replacement_bios.php

Yes, all I did was swap the rom version. Keep in mind that you also need a dual oscillator mod if you aren't using RGB. I am personally using a ps1digital which has one built in.

By the way, this doesn't fix the toy story 2 and wipeout 3 checks which do seem to be based on mechacon and not on the rom version.

As to how it runs on PSIO, it can't unless PSIO patches the check in the executable's memory.

AndreasV81, 2021-03-29:
Why did you swap the bios after all? Were there any issues you had with other titles? Or just for modding and personalizing the ps1?

mathieulh, 2021-03-29:
> Why did you swap the bios after all? Were there any issues you had with other titles? Or just for modding and personalizing the ps1?

I just wanted to have the BIOS menu and the boot screen at 60hz and where I live (France), sourcing a PAL PS1 is easier and cheaper.

AndreasV81, 2021-03-29:
Ok, i dont think i will do that for just one game. Still thanks for the hint, what you discovered is that it should be indeed possible to patch the game as cybdyn did something that worked and it is not a mechacon issue. Which ps1 model and mobo did you mod with the bios?

mathieulh, 2021-03-29:
> Ok, i dont think i will do that for just one game. Still thanks for the hint, what you discovered is that it should be indeed possible to patch the game as cybdyn did something that worked and it is not a mechacon issue. Which ps1 model and mobo did you mod with the bios?

I am using a PAL PU-18 board with a US PU-18 BIOS.

AndreasV81, 2021-03-29:
I will try next to emulate the game with no$cash disassembler. Maybe dumping the ram when emulated with different bioses could give a hint what is done differently. But i dont think it will be easy 😅😅😅

danilodequeiroz, 2021-04-04:
Is the xstation compatible with NTSC-J PS1 no matter what BIOS they are using or been modded? I have one 5500 modded with a NTSC-U Bios and a 5501 with a possibly faulty IC 704 (CD Driver IC) and also been looking for Japanese consoles since they are easier to import here and they are usually in good shape internally and externally.

ramapcsx2, 2021-04-04:
As you add more modifications, it becomes harder to tell.
It should work, but specific games that use region lockout code may require the "correct" BIOS or mechacon chips to work.

danilodequeiroz, 2021-04-06:
> As you add more modifications, it becomes harder to tell.
> It should work, but specific games that use region lockout code may require the "correct" BIOS or mechacon chips to work.

Do you mean these games https://consolecopyworld.com/psx/psx_protected_games_jap.shtml

ramapcsx2, 2021-04-06:
No, not necessarily those. Anti-mod is not an issue for xStation.
Which games are using additional region checks is still relatively unknown today. We just find them as incompatibilities over time.
WipeOut 3 played on a PAL mechacon is one such game, Toy Story 2 is another.
Soul Reaver NTSC-U is linked to the BIOS region in comparison.

mathieulh, 2021-04-06:
> Is the xstation compatible with NTSC-J PS1 no matter what BIOS they are using or been modded? I have one 5500 modded with a NTSC-U BIOS and a 5501 with a possibly faulty IC 704 (CD Driver IC) and [...]

Since it's specifically checking the BIOS region, the 5500 running a NTSC-U BIOS should work, I am not sure if the Japanese BIOS actually passes the check though, the PAL does not, obviously.

AndreasV81, 2021-04-06:
As far as i know the game works on japanese ps1s.

mathieulh, 2021-04-08:
> As far as i know the game works on japanese ps1s.

Yes, it does, the check appears to specifically verify that the BIOS address does not have the 'E' string.

mathieulh, 2021-04-13:
I may have found the check in the NTSC-U version of soul reaver: 

`TEXT:800361BC                 slti    $v0, $a0, 0x45  # 'E'
TEXT:800361C0                 bnez    $v0, loc_800361DC
TEXT:800361C4                 li      $a2, 0x400
TEXT:800361C8                 addiu   $v0, $a0, -0x44
TEXT:800361CC                 srl     $v1, $v0, 31
TEXT:800361D0                 addu    $v0, $v1
TEXT:800361D4                 sra     $v0, 1
TEXT:800361D8                 addiu   $a0, $v0, 0x44  # 'D'`

This would make the location at 0x3A1C4 in the ISO, where you would want to change 0x45 to a value that's not a valid BIOS region string (like 0x00 for example)

Unfortunately I cannot test this patch since I am already using a NTSC-U BIOS.

mathieulh, 2021-04-14:
So, someone with an unmodded console attempted to reproduce the freeze, with and without my patch and couldn't.

Do you have a reliable trigger for it? He tried the door in the drowned abbey, that's the right symbol to teleport to, right?
[image]

mathieulh, 2021-04-14:
Ok, I reproduced the crash using this symbol even with NTSC-U BIOS

https://imgur.com/orfRdxd

It seems to indeed be a mechacon check, not a BIOS one.

p-b-p, 2024-02-08:
Here some more info regarding this crash:

Tested with Legacy of Kain: Soul Reaver NTSC-U (SLUS-00708 with verified redump md5):

1. PAL setup: SCPH-5552 on xStation with no other modifications - the crash is reproducible as expected.
2. PAL setup: SCPH-9002 on CD-ROM with MM3 modchip and US BIOS installed (SCPH-9001 Version 4.1 12/16/97 A) - the crash is reproducible.
3. PAL setup: SCPH-7002 on CD-ROM with UniROM 8.0.K installed on Pro Action Replay 2 - tested it many times and I wasn't able to reproduce this crash.

Was it possible that UniROM handles differently a mechacon (MC68HC05) checks?

ramapcsx2, 2024-02-08:
Just as an information, games have an easy time telling the Mech and the BIOS chip region. You may assume that a region lockout routine has both chip's region strings and can use those as wanted (or demanded by regulation via Sony regional stuff).
Additionally, there are games known to check for the presence of expansion ROM carts, one example being an NTSC-J Cardcaptor Sakura game, which does not boot if any ROM is attached, no matter the contents of that ROM.

Now add to that the coder's likely desire to hide their region checks, and it becomes really hard to guesstimate what goes wrong.

At this point, someone really needs to dissect the binary, otherwise we'll never know :p

p-b-p, 2024-02-08:
Yes, but is still interesting why it works with PSIO and UniROM, but not on real CD-ROM or XStation. Maybe UniROM behave similarly as PSIO like mentioned [here](https://github.com/x-station/xstation-issues/issues/93#issuecomment-798555235)

So, in order to investigate further I've ordered a spare PU-18 NTSC-U board and the next test I plan is to replace the mechacon on the the XStation console and see if that will resolve this issue :)

AndreasV81, 2024-02-08:
Thank you very much for diving into a 2 year old issue :) It would be really great if this could be resolved somehow.

p-b-p, 2024-02-10:
I've played for a while with no$psx debugger and found a very interesting mark about this check :)
Standing right before the door the game on the real hardware hangs. When I simulate in on no$psx you can see on the image below that CDROM sends 3xGetID commands... I asume that here things get broken because it receives 'E' from the PAL mech, but expects 'A' to continue. However, receiving 'I' also pass the check because as reported NTSC-J consoles don't have this issue.

[image]

Apart from that, back in the days when I played that title on my modded SCPH-9002 PAL console I'm pretty sure that the disk was NTSC-U version of the game and I don't remember to had such random hanging issues during the full gameplay. Guess I have to thank to my dump modchip which was constantly sending 'SCEE' string when the console is turned on :D

AndreasV81, 2024-02-10:
So the correct modchip would solve this issue?

p-b-p, 2024-02-10:
> So the correct modchip would solve this issue?

Probably :) I have to test it. Unfortunately I've sold the console long time ago and don't remember what exactly was flashed on that chip. Only remember that it was 4 wire mod wired as shown on here:

[image]

[37 more replies]