x-station/xstation-issues #94: Undocumented PU-8 (21) revision?

WigglesEX, 2021-03-13:
Started an Xstation install this morning on a SCPH-1001 Playstation with a PU-8 (21) board. Everything seemed fine until I got to the part where you're supposed to solder a wire from the test pad in the QSB to the point outlined in the instructions. My late model PU-8 does not have the same configuration as the one in the picture or really like any I've done so far (this is my third PU-8 Xstation install). 

There is a point roughly in the same spot as the one in the install instructions so I soldered my wire there, but now have a blank screen when I power the PS1 on (no Sony screen, no Playstation screen, no Xstation menu, nothing). I've gone back and checked my work, tested solder joints for continuity, resoldered lifted pin pads and removed resister sites, but still nothing. The only thing I can think of is my board isn't supported despite being a PU-8. I'll continue to troubleshoot, but thought I'd see if you think my weird board revision is the reason why it's not working.

Thanks

[image]

WigglesEX, 2021-03-13:
Well I answered my own question; yes this weird PU-8 variant is supported, just be careful that no unsheathed wire is touching the adjacent resistor. That's what was causing it not to boot.

closed 2021-03-13 by WigglesEX

ramapcsx2, 2021-03-14:
If the QSB fits, the model is supported :)