x-station/xstation-issues #95: PS1 not turning on, after xStation install

chameleon2606, 2021-03-16:
I was really exited to install my new xStation today.
Installation went smoothly.
I had no problem lifting the pins and soldering the QSB to the board and I double checked all the solder points with a multimeter.

[image]

[image]

[image]

[image]

Upon reassembling the PS1 (scph 5552) i tried to power it on.. but i wouldnt, as if the power plug wasn't connected.
The power light from the console doesn't light up and the xStation biefly flashes blue.
When I disconnect the power supply from the mainboard then the green light will turn on again.

[image]

I tried connecting the board to another ps1 psu, but the problem persists.

I'm really confused, as I haven't seen anyone else with this issue.

closed 2021-03-16 by chameleon2606

ramapcsx2, 2021-03-17:
Hey, I noticed that you closed the issue.
If you need help here, let me know.

(There's a short clearly visible. You probably noticed :p)

chameleon2606, 2021-03-17:
> Hey, I noticed that you closed the issue.
> If you need help here, let me know.
> 
> (There's a short clearly visible. You probably noticed :p)

exactly.. as soon as I posted the image, I realized that.

it's just weird cause I've never touched that resistor.

but everything works perfectly now!