x-station/xstation-issues #96: xstation will not send game id when set to run last game on power up.

killbot80k, 2021-03-17:
I have an xstation along with a PS1digital and I have been using the per game settings. For this to work it requires the xstation to send the game id to the PS1digital and typically works great. However, when I set the xstation to boot to the last game played it never sends the game id so the PS1digital will end up using the global settings.

ramapcsx2, 2021-03-17:
Ah yeah, I can see why. The game id can only get sent from the xStation loader. If the loader doesn't run first, it won't work.

Maybe this is something that the ps1digital firmware could fix..
I'm on it!

ramapcsx2, 2021-03-17:
Chriz is on it :)

ramapcsx2, 2021-03-22:
I'll close this, as it's been taken over by ps1digital devs ;p

closed 2021-03-22 by ramapcsx2