x-station/xstation-issues #97: Bug with multidisc mode in the loader from the release 1.1.0

elssbbboy, 2021-03-17:
Hello, I received, installed and updated my xstation to the version 1.1.0 a few days ago. I detected a problem when using multidisc mode (tried with Chrono Cross and Xenogears from several sources). When selecting both fast or full boot, it sometimes doesn't boot. It just freezes in the screen (xstation and the title of the game), it doesn't show "loaded system.cnf from...", and the green led keeps blinking, and it doesn't happen when selecting just one game from the list.

At first, I thought it was my fault with a bad installation, but I checked everything with a multimeter and I didn't find any problems with it. I also reformatted my microSD card and tried with a faster different card. It resulted into the same problem.

So after all that testing, I tried to use the firmware from the release 1.1.0 and the loader from the release 1.0.5. Finally, that solved my issue.

ramapcsx2, 2021-03-18:
Hey,
a new firmware + loader are currently in testing. It should see a release in 2-3 days.
It'd be great if you could try again with that update :)

elssbbboy, 2021-03-19:
Thank you for your reply. As soon as it is released, I will try it :-)

carmax91, 2021-03-19:
I have the same issues with the same games, the problem was with sd extender

ramapcsx2, 2021-03-22:
Hey,
the fixed firmware + loader are out. Could you verify that this all works now please? :)

elssbbboy, 2021-03-23:
Hello, I installed the firmware and the loader from the release 1.2.0 and I can confirm it works properly now. Thank you for your time.

ramapcsx2, 2021-03-23:
Cheers :)

closed 2021-03-23 by ramapcsx2