x-station/xstation-issues #98: Toca 2 Touring cars - No music at all

boquetipo, 2021-03-18:
Hi, I've just tested the game Toca 2, and I got no music whatsoever. Game works fine, and I also can hear sounds (menu, engines, crashes, etc.).

Though yesterday was the first day I tried xStation, I got music from other games.

ID game: SLES_015.42
PSX: 5502 PAL
xStation firmware: 1.1.0 (updated on 2021-03-17)
A/V cable: RGB

Cheers.

boquetipo, 2021-03-18:
And PU-18 board

ramapcsx2, 2021-03-18:
The most likely cause is a bad rip with missing or bad cue sheet.
Is this a known good rip?

boquetipo, 2021-03-18:
Sorry, my bad.

Not a good rip. Same behaviour on emulator.

I will pay attention from now on, and check thoroughly next time I decide to waste your time, guys.

Cheers.

closed 2021-03-18 by boquetipo