x-station/xstation-issues #99: Xstation didn´t load boot menu -->ramapcsx2 may help?

TrueCrashB, 2021-03-19:
Hi
i have no idea why it didn´t load the boot menu.
Ok, i have no soldering skills, but i thought i did id fine.

I read this in another issue:

ramapcsx2:
"Are you located in Europe, or otherwise willing/able to send me the whole board + xStation?
I'll fix it ;p"

Dear ramapcsx2 could you have a look on it? 
I wil send you the board + xstation.

ramapcsx2, 2021-03-19:
Hey,
don't you want to try debugging it first?
The other issue, it became clear from the pictures and the chat that he wouldn't be able to fix it himself.

TrueCrashB, 2021-03-19:
You are right!
Here are the pictures.
[image]
[image]
[image]
[image]

Thx

ramapcsx2, 2021-03-20:
I can hardly see the important details, but 2 pins on the DSP look like they are connecting with each other.
The QSB has huge solder blobs standing up. Some of them might even make contact with the bottom shield > would cause a short.

Do you get a blue power light on the xStation, and is the at least a short green LED flash when it powers up?

TrueCrashB, 2021-03-20:
Hi thx for your answer. Ok, i seperated the pins on the DSP, but i am not sure if they had contact. Ok, nevertheless now they don´t have contact. I try to correct the solder blobs, but i am afraid to damage something. 
Yes, i got a steady blue light and if i press the lid button a green flashes once and than the double.

bahamutfan64, 2021-03-21:
Like Rama said, the big solder blobs may be an issue - all you need to do is **add flux** to them and with a clean tip go over them once more. Big blobs like that can basically act as cold joints, so all we're doing is reflowing them to ensure they're making the proper connection with the appropriate points below the QSB. 

Clean your tip after every blob and do the same thing for each one.

ramapcsx2, 2021-03-22:
Right, it sounds like you're missing at most 1 or 2 signals on the QSB (meaning they're not properly connected to the PSX).

TrueCrashB, 2021-03-26:
Hi, ok i resoldered all points and hey, it seemed to work.
I got the XStation Screen and it says "No Games found. Press (Triangle?) to refresh list!
I made the update which worked fine.

Now it wont load the XStation menu any more.

ramapcsx2, 2021-03-26:
Someone else had a similar issue, where the older firmware worked and the newer did not.
It was another open connection for him. Soldering it down fixed it.

TrueCrashB, 2021-03-26:
Ok, that means to resolder all points again? Ok, next try. Thx.
I will keep you up to date.

TrueCrashB, 2021-03-30:
Now i am at the end of my capabilities. I soldered every point used a lot of flux, but it didn´t work. I got a steady blue light. I saw the green light which might indicate readings/writing at the card. But i won´t load the menu. I attached some pictures of my soldering.
I don´t use an SDCard expander.
![Uploading 20210330_205124.jpg…]
[image]
[image]

[image]
[image]
[image]

ramapcsx2, 2021-03-31:
So you want to send it to someone?
I can fix it for you, but you've got to handle shipping back and forth. I'm in Germany ;p

TrueCrashB, 2021-03-31:
> 
> 
> So you want to send it to someone?
> I can fix it for you, but you've got to handle shipping back and forth. I'm in Germany ;p

I would be really grateful if you could fix it.
I don´t know what else to do to fix it.
I´m in Germany too.
I'll pay all shipping costs and if there are additional costs for parts, of course.

ramapcsx2, 2021-04-01:
Can you drop by on Discord and send me a PM?
https://discord.gg/QUzrgUYs

TrueCrashB, 2021-04-14:
Thx for the help.

closed 2021-04-14 by TrueCrashB

ramapcsx2, 2021-04-14:
Ah yea, a mass of solder crept under the QSB and bridged points there.