x-station/xstation-issues #100: SuperLite 1500 Series - The Tetris (Japan) cannot load.

MythicWebsite, 2021-03-21:
I've ripped the game from the disc myself and it works on emulator. I've also tried it with a version I found online and neither one would load on the console. You select the game and it just sits there until it throws an error.

ramapcsx2, 2021-03-21:
This game? https://i.imgur.com/jw7sJYh.jpg

MythicWebsite, 2021-03-21:
That would be the one. It's the only game so far that I have been unable to get to work. While it's not the greatest game, I do own a copy and would like to get it working on my console.

MythicWebsite, 2021-03-21:
It hangs on this screen https://i.imgur.com/pfHugUX.png
Then it gives this error after a long wait https://i.imgur.com/EK8rO5K.png
If it gives that error, then it gives that error for every game afterwards until I turn off the console https://i.imgur.com/PA6gSHm.png

blzla, 2021-03-21:
This sounds like a firmware issue that was reported on Discord a couple days ago. You could test if it also happens with firmware 1.0.5 or wait for the release of 1.2.0, which should be soon. There's also an open beta of 1.2.0 RC2 available on Discord that fixed the issue.

MythicWebsite, 2021-03-21:
Is there a link anywhere to the discord so I may try the beta file?

blzla, 2021-03-21:
It's in the Classic Gaming Discord: https://top.gg/servers/418895913210216448

MythicWebsite, 2021-03-21:
The beta did fix the game not being able to load. Thanks for your help!

closed 2021-03-21 by MythicWebsite