x-station/xstation-issues #101: Game list is showing hidden files

MrBlank, 2021-03-24:
XStation Firmware + Loader [1.2.0] - 2021-03-22

Looks like this firmware update is finding hidden "dot" Mac files when searching for games and adding them to the list of games. Reverting to the previous firmware fixes this.

ramapcsx2, 2021-03-24:
Yeah, I heard from someone else as well.
I'll have to extend that code to work with subfolders as well. Will do that! :)

MrBlank, 2021-03-24:
Super. I love the xStation. Excited to see what else it can do.

ramapcsx2, 2021-04-07:
The bug is fixed here and the fix will roll out with the next update. I'm closing this, because I'll forget to do it later..
(In case the next update doesn't fix it fully, feel free to create a new issue. But it should be fine :) )

closed 2021-04-07 by ramapcsx2