x-station/xstation-issues #103: Xstation loads last game instead of menu - multi-disk

kimbapslice, 2021-03-31:
scph-5501, ps1digital 1.5.2, xstation 1.2.0. 

Start a multi-disk game (fast load), once the game begins, turn off the system, then open the lid, power the system on, at the bios, now close the lid. The game will now load instead of the xstation menu. Xstation setting is to go to loader (not last game). Power cycling with the lid close will get back into the menu.

ramapcsx2, 2021-03-31:
Yep, that sounds good :)
XStation is focused on what happens after you do your selection in the loader.
It should behave very much like a real drive then, swapping through the queue or simply retrying the current disc, in case a game hung.

You want the secondary function: Mount the loader and get back to the menu.
This is achieved by pressing the reset button for a short while (1 second).
You get there

kimbapslice, 2021-03-31:
Fantastic! I didn't know if it was a bug or the intended flow, figured I would report it anyway :) Thanks Rama!

closed 2021-03-31 by kimbapslice