x-station/xstation-issues #105: Vagrant's Story NTSC-U won't boot - fw 1.2.0

ABelliqueux, 2021-04-05:
As title says, Vagrant's Story NTSC-U won't boot.
If launching from the menu, it freezes just after the xstation loading screen ( with system.cnf mention)
If hitting reset, the psx just boots to bios.

The backup is from the Redump archive : https://archive.org/details/Redump.orgSonyPlayStation-NTSC-U-V

ramapcsx2, 2021-04-05:
Works perfectly fine here. Tell me more about your setup, PSX region and such.

ABelliqueux, 2021-04-06:
Hi, I've updated my previous message with more details.

ramapcsx2, 2021-04-06:
It could be a yet unknown region lock on the PAL console.
I'll test this out.

ramapcsx2, 2021-04-06:
It works on my pure PAL machine as well.
Here is a test for you: Please try to move the folder to another place on your SD card.
It might have trouble building the path name correctly.

ABelliqueux, 2021-04-06:
Well I tried renaming it and moving it to the root of the file tree but same result.

Also, I noticed when checking in `01xstation/games.txt`  and `gamescache.txt`, xstation insists on putting Vagrant Story at the end when refreshing the list :  

```
/U/Um Jammer Lammy [U] [SCUS-94448]/Um Jammer Lammy [U] [SCUS-94448].bin
/U/Urban Chaos (USA)/Urban Chaos (USA).bin
/VZ/Vib-Ribbon (Europe) (En,Fr,De,Es,It)/Vib-Ribbon (Europe) (En,Fr,De,Es,It) (Track 1).bin
/VZ/Wipeout 3 [U] [SLUS-00865]/Wipeout 3 [U] [SLUS-00865].bin
/VZ/YAROZE Games-PSX-PAL&NTSC/yaroze.bin
/VZ/Worms Armageddon (USA)/Worms Armageddon (USA).bin
/VZ/Vagrant Story/Vagrant Story.bin
```
Strange...

ramapcsx2, 2021-04-06:
Not really, no.
The file gamescache.txt stores the individual games as they are detected. If you copied the game last, then it will show up last in the list, too. (So that's all fine.)

Does Worms Armageddon work? Asking because the game before that has the ampersand & character in it. Not sure about that.

ramapcsx2, 2021-04-06:
Oh and you did a game list rescan after you moved it, right?

ABelliqueux, 2021-04-06:
Yep, did several rescan. Worms and VibRibbon work fine, I have not tested the others yet.
In the meantime, I removed and made a new copy of the cue/bin  on the sdcard, to no avail...
Next step is trying with a backup from a different source..

ramapcsx2, 2021-04-06:
Okay, great.
Please also check that you're not using any 3rd party tools (like Unirom).
A basic console with just the xstation, power up mode set to "boot last mounted game", then power up and let the BIOS shell run the game (if it can).

mathieulh, 2021-04-06:
I just tried it here on my PAL SCPH-5502 (modded with US BIOS) and it works/boots fine here using the redump bin/cue.

ramapcsx2, 2021-04-07:
So this was a bad dump in the end. Not sure how that happened, but the game works fine :)

closed 2021-04-07 by ramapcsx2