x-station/xstation-issues #106: Won't boot up

lapinuelle, 2021-04-05:
Hi! After installation I stuck at MC/CD screen. All pins connectivity is OK, but smhw pin8 of qsb is connected to ground (pad on MB especially). Goy stable blue light on main board and 2 quick flashes of green on startup

ramapcsx2, 2021-04-05:
Not sure what to tell you. You need to debug more :)
The installation is not complete, the switches might not work well, etc..
If you need help, you can always send the console to an installer.

lapinuelle, 2021-04-06:
Well, I understand, that something is not okay :) does pin 8 should be connected to ground or it is an issue with my mb?

Отправлено с iPhone

lapinuelle, 2021-04-06:
Also, the lid switch is just bridged on my console (was done before me). MB model is PU-18.
Previously I had install psnee and it worked like a charm. With XStation I had stuck on mc/cd screen with no response from gamepad

Btw, are there spare qsbs available anywhere? 

Отправлено с iPhone

ramapcsx2, 2021-04-06:
Which pin 8 is this? Can you link to the picture you mean?
A bridged lid switch is fine for now, but not ideal. You won't be able to switch game discs if a game requires it.

Spare QSBs here:
https://castlemaniagames.com/products/xstation-qsb

lapinuelle, 2021-04-06:
Отправлено с iPhone

bahamutfan64, 2021-04-06:
You’ll have to show us a pic of your qsb installation.

ramapcsx2, 2021-04-06:
This pin 8 is the connection to the lid switch. Since you've soldered it down to ground, it should be grounded.

lapinuelle, 2021-04-06:
All points are connected, checked it triple time. After startup and hanging at Mc/cd screen there is a config.txt on sad card with only one symbol inside. 
I've botched some pads but restore them with wires

lapinuelle, 2021-04-06:
[image]
[image]
[image]

Yeah, I know, laugh on me, shame on me

lapinuelle, 2021-04-06:
Remembered now, that before, when I had chip installed (Noname one, then replaced by psnee) my console booted to game not from first try. And it was not drive issue, cause same drive worked like a charm in slim console. I start to think, that all my problems caused by little bit faulty motherboard.

ramapcsx2, 2021-04-06:
Well... The problem is certainly with the QSB, any of the points it connects to, etc. The entire area is suspect.
Check that none of the QSB solder points contact each other. Check that each of the points reaches the flat flex connector still.

I can't really recommend much. An installer would use the scope now, but I don't think you have one nearby? :p

lapinuelle, 2021-04-07:
All connectivity is ok, but it still going directly to MC/CD screen, without any delay, just like the lid is opened. But the switch is soldered to be permanently "as closed". 
I think I ether have faulty motherboard or screwed up installation.

xStation board still have solid blue light, and two green flashes at startup

Maybe, i will try to find another console (in Russia it is not so easy, most of them 9000s or slim).
Btw, are there any sellers of QSB ir Europe? Catlemania is located in USA, shipping time and cost will be not so good

ramapcsx2, 2021-04-07:
Oh, that's bad..
You have several potential issues, and ideally someone with a scope should help you out now.
All I can tell for sure is that the xStation itself is fine. A couple green LED flashes on bootup is good.

This installation pictures look like you could have easily damaged a PSX mainboard trace.
The board could also be bad, as you say.
I don't know whether you managed to follow each solder point to the destination chip (mechacon and 1815Q).

lapinuelle, 2021-04-08:
I ordered another console so will try not to screw that one.

closed 2021-04-08 by lapinuelle

ramapcsx2, 2021-04-08:
Good luck with it!

Today I've helped fix an install where the QSB "IO4" point was making contact with the large test pad right below the QSB:
https://i.imgur.com/z7jRc4z.jpg
The via at IO4 should not have contact to the solder blob below it.

lapinuelle, 2021-04-12:
Rejoice for me! I've got another console and with some "Spartan" installation (soldering each pin of ribbon connector on pu-8 qsb with single wire to the corresponding pad on pu-18 mb, lol) I've got it working. Thank you for great piece of hardware and all help that I've got from you during this process