x-station/xstation-issues #108: Xstation won't boot

Pringles36, 2021-04-12:
Hey everyone. I have a 5502 (PU-18) Playstation that I've just attempted to install the Xstation as per the instructions. After completion i have gone to boot my Playstation and only get a black screen and no video. The startup audio does play but half way through, before the transition into the higher "ping sound" the track just loops at the midway point. I'm not sure where ive gone wrong.  Could you please check my work. The Xstation displays a solid blue and i get 1 to 2 flickers of the green led. And i have also checked all the test points with no discernable issues.
I, uhh, awhile ago also did some prep work for when i get the PS1 digital that i was wondering if that would cause any issues. I completed step 1 and 2 of this guide http://dansprojects.com/PS1Digital_install.html , would removing the components indicated cause any compatibility issue to the Xstation?
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

ramapcsx2, 2021-04-12:
Yeah, is the GPU oscillator removed in the ps1digital install step?
You need that to boot ;p

Pringles36, 2021-04-12:
Hahah! In the install guide for the PS1 digital it just says "remove surface mount component". So, is that what i removed?? Have i damaged anything by trying to boot without it or can i go ahead and install the PS1 digital and pray everything works fine after that?? Can you see any issues with the soldering/pin lift etc? Thanks heaps for your quick reply by the way!

Pringles36, 2021-04-12:
[image]
I found this. So yeah i think you're right. So, yeah, have i done any damage that needs to be rectified or can i install the PS1 digital and everything should theoretically work?

ramapcsx2, 2021-04-12:
Nah, you won't damage anything right away.
Simply continue the install and it should be alright.

Pringles36, 2021-05-10:
> Any updates?

Yeah sorry, I should post an update! So, I got impatient waiting for the PS1 digital to arrive. So what I did is I desoldered the Xstation QSB and installed it on to another unmodified board. Fired it up and it works perfectly. So now just more waiting for the PS1 digital to arrive to complete the unit. :)

ramapcsx2, 2021-05-10:
In this case, it was probably damage to one of the PSX chips then. But nice you got it working :)