x-station/xstation-issues #109: Marvel vs Capcom stuttering audio after awhile

GabeShack, 2021-04-15:
Using xstation with PS1 digital and a laser bear micro SD to SD extension
Noticing stuttering audio (gameplay is unaffected) after playing marvel vs Capcom for 20-30 minutes. 
Using a Lexar 1TB Professional 633x UHS-I SDXC Memory Card.

ramapcsx2, 2021-04-16:
Hey,
please try a different SD card. The Lexars are known to have a fault where they become too slow to be used after a while.

GabeShack, 2021-04-16:
Will do. This is however a brand new card.

GabeShack, 2021-04-16:
Also interesting that only the audio gets affected by the slow down. Game play otherwise isn’t altered.

ramapcsx2, 2021-04-16:
The cards have a faulty controller that simply stalls after a while. It can be recreated on a computer with a good micro sd adapter.

ramapcsx2, 2021-04-16:
You may also notice that it recovers after a while.
Music should instantly play normally again then.

GabeShack, 2021-04-17:
That is correct. It does recover. Sound effects still work normally too even when the music is stuttering. On another Lexar 1tb card I’m not seeing the issue, but I’m betting it will happen if I fill the card like the 1st one.

Not a big issue. Thanks for taking the time.

closed 2021-04-17 by GabeShack

ramapcsx2, 2021-04-17:
Okay. Sorry about the situation ><
It's weird, but even really old cards are usually fine with the requirements for xStation.
I've got old 4GB class 4 cards here that can do it, but the Lexar 633x 512GB card goes to have tea time regularly.
Another one is an Evo+ (not "plus") from Samsung, an orange 128GB model. Yet another Samsung is fine again.