x-station/xstation-issues #110: Xsatation  240Psuite won´t boot

RetroFanFran, 2021-04-19:
I have tried launching Artemio Urbina's 240P test suite for Ps1 (cue + bin files) and the Xstation freezes and won't load.

My console is a SCPH5502, PU18 PAL, and my Xsation is on latest firmware and works fine on everything else.

I know it is not a game, but it is an interesting enough tool to try to be supported by the ODE

ramapcsx2, 2021-04-19:
Please check here: https://github.com/x-station/xstation-issues/issues/21

RetroFanFran, 2021-04-19:
Great! Thanks! but i still have the issue and the suite won´t work on my system
> Please check here: #21
The image of this thread is gone 

> or use this image. Both should be correct:
> [filipalac/240pTestSuite-PS1#12](https://github.com/filipalac/240pTestSuite-PS1/issues/12)
I have used this one but ... I boot it up, PS boot and ps logo appear at first like in a game and after that the console stops sending any images to my TV and I have to reboot the system

It is a small advance since before the ODE was frozen, but it still does not start the suite ...

ramapcsx2, 2021-04-19:
Does your console run the xStation loader and other games fine?

RetroFanFran, 2021-04-19:
Yes, all the games that I have tried work correctly ..., I installed the ODE on January 2 and after many hours of play I can confirm that my xstation works fine ...

As for the 240psuite, in addition to the one you link to me, I have also tried without success the standard distribution and the distribution for emulators .... with the same result, one freezes the xstation, and the other suppresses the video signal to my tv

ramapcsx2, 2021-04-19:
Your console is a PAL, right? Can you play NTSC games on it fine?
(I wonder whether you have a rare unit that can't display the out of region video mode.)

RetroFanFran, 2021-04-24:
> Your console is a PAL, right? Can you play NTSC games on it fine?
> (I wonder whether you have a rare unit that can't display the out of region video mode.)

Yes, my console play NTSC-usa, NTSC-japan and PAL games without any issue...

ramapcsx2, 2021-04-24:
Curious.
If you want, you could try the 1.3 beta over at Discord.
https://discord.com/channels/418895913210216448/802566037815427102
It's pinned in the channel.

But I can't explain it. If other games work, a fixed 240p test suite image should also work..

ramapcsx2, 2021-05-09:
Any updates here?

RetroFanFran, 2021-05-09:
nothing new, still won´t work. This past week i install two ODE´s in two new consoles, and try again with the same result.... as i expect isn´t a particular issue with my personal system

birdybro, 2021-05-10:
Just chiming in with my own testing and files attached so hopefully you can find one that works for you... For mine, on Firmware 1.2.0 (Mar 21 2021), this is the following behavior:

[240pTestSuitePS1.zip](https://github.com/x-station/xstation-issues/files/6448919/240pTestSuitePS1.zip) <-- This one **doesn't** work.
That's CRC32 Hash 7092C879 for the bin, the official one --> https://github.com/filipalac/240pTestSuite-PS1/releases/download/19122020/240pTestSuitePS1.zip

[240pTestSuite-rama.zip](https://github.com/x-station/xstation-issues/files/6448920/240pTestSuite-rama.zip) <-- This one **does** work.
That's CRC32 Hash E150882F for the bin. This is from the issue that @ramapcsx2 raised here --> https://github.com/filipalac/240pTestSuite-PS1/issues/12 - @filipalac just hasn't updated the image yet in response to the issue Rama raised.

If the second one doesn't work for you, then your system is behaving differently than my xStation installed PS1 is.