x-station/xstation-issues #112: Video issues after install

SilverBolt76, 2021-04-21:
Not SD card related appears in boot as pictured.  Any ideas?  PS worked fine before installation.  Xstation boots fine and boots games fine.  Tried several games.
[image]
[image]

ramapcsx2, 2021-04-21:
It looks like a GPU or GPU RAM chip issue. It might have developed when installing the xStation.

If you want to try debugging it, try running the system with the shell and shield removed, then press down the GPU are chips lightly (but mind the power supply!).
If you can affect the look of the glitches by pressing down on some chips, you know it's cold solder joints.

birdybro, 2021-04-21:
Agreed, the most likely explanation is that upon install you unknowingly flexed the PCB in some places and the solder joints got tiny cracks in them.

SilverBolt76, 2021-04-21:
I will remove the shield and check the board.  Thank you, gentlemen.

closed 2021-04-21 by SilverBolt76