x-station/xstation-issues #113: Not Working after firmware update?

Evangeliman, 2021-04-29:
So after a putting the mod in my 5501 ps1 pu-18 I finally got it to boot after double checking my soldering and pins, it went into the menu, and then I installed the latest firmware (before doing anything else) and when the system automatically rebooted it went to the the memory card/cd player menu. holding reset for second or 2 and resetting the console via power button doesn't make the launcher appear. I dunno what to do... any reason it would boot once and then never again?

ramapcsx2, 2021-04-30:
Hey,
you probably still have soldering related issues.
A signal that has intermittent contact might result in a successful boot on one firmware, but not on another.
Reflow all your points again please.

Evangeliman, 2021-04-30:
Ok, so i opened it up and rechecked everything and made sure everything was soldered down, I found that pin 57 is gone, just ripped off, it must have happened in when i was lifting pin 56. Any idea what pin 57 does or if i can salvage the system somehow?

ramapcsx2, 2021-04-30:
Pin 57 on the DSP does nothing in the PSX. It's okay if it's broken off.
Keep searching :)

ramapcsx2, 2021-04-30:
Also, if the problem remains, what happens when you go into the CD player in the BIOS?
You may want to open the lid to enter the player first, then close it.

Do you see a single track of ~30 seconds then?

Evangeliman, 2021-04-30:
No cd player just says 0 - blank min - blank sec, though sometimes it says 0- 0 -0, after i opened it up the first time to check everything, it booted into the menu once more, i tried to reset game list, (just one test game) it froze on retrieving game list after a manual restart it just booted back into bios menu. cant get it to boot launcher.

ramapcsx2, 2021-04-30:
Okay, this is definitely a cold solder joint or an intermittent short.
It can be very difficult to find and fix these.
If you find that nothing you do helps, consider sending the entire console to an installer.

Evangeliman, 2021-04-30:
yeah think i will just desolder the qsb tomorrow and try again. the pins looked fine but ill have to double check. thanks for the help btw.

ramapcsx2, 2021-05-09:
Any updates? 
Consider giving the job to an installer :)