x-station/xstation-issues #114: No power to xStation

OhPoorPup, 2021-04-30:
Hi, 

I had my xStation all set up and working for about two months with no issue, but I think it must've gotten moved around improperly because it began to demonstrate the "full load/reset" issue listed in the readme. So I desoldered the QSB, checked for proper alignment, used plenty of flux, checked continuity when I was done and powered it up. Success! Everything worked normally. I put the lid back on and put it back in my entertainment center, and no video signal. Open it up and the xStation isn't receiving power, though the PS1Digital appears to be working fine. (I know if the "CD drive" isn't detected a video signal won't display.) I plugged in the power supply at my workbench to test and a fuse on the PSU blew.... replaced it with another PSU and it seems to power on fine now.

I've reworked the joints on the QSB, tested continuity, recapped the CD drive region of the board, made sure all the resistors on the xStation board aren't open, checked the fuses over in the bottom-left region of the PU-18 board and checked for continuity on the ribbon cable and I'm just at a loss. I saw ramapcsx2 suggest on Twitter to check if the disc drive itself spins up on power and it does. Are there any troubleshooting suggestions for driving power to the xStation board specifically? Which solder point is delivering power through the QSB? Here's some pictures of the QSB as it stands right now:
[image]
[image]
[image]
[image]

Thanks for your help, I really love this device and would love to get it working again.

ramapcsx2, 2021-04-30:
Hey,
so the problem right now is that there's no blue LED?
Power to the xStation is sourced at the 7.6V point to the right of the BA6392FP chip.
You should be able to measure continuity from there to the power supply, and with the xStation connected, also to the 3 pin LDO on it.

OhPoorPup, 2021-05-01:
Yeah that's the issue. I wasn't getting continuity from pin two on the QSB to pin two on the xStation, so I removed the ribbon connector and replaced it, I'm getting full continuity now but no power LEDs.

ramapcsx2, 2021-05-01:
Seems all pretty bad. Not sure what I can do from here. Keep debugging! :p

OhPoorPup, 2021-05-01:
I fixed it! It looks like something was messed up on pin 2 of the xStation's ribbon connector so I replaced it and all good now. thanks!

ramapcsx2, 2021-05-01:
Fingers crossed it stays that way :)

closed 2021-05-04 by OhPoorPup