x-station/xstation-issues #115: R4 Ridge Racer Type 4 Broken after latest update

samson7point1, 2021-05-03:
Severe graphic issues in R4 since the latest update - particularly during opening FMV but also on menus.  Opening FMV begins in the wrong position and is split in half horizontally so that the top half plays at the bottom of the screen and the bottom half plays at the top with a large gap in between.  Will upload video later on - just would like someone to check first if this is a common/known problem.

OhPoorPup, 2021-05-04:
Do you have a PS1Digital attached? I believe this is an issue with its Auto horizontal/vertical adjustments, try setting horizontal and vertical to "0" instead of "auto"

samson7point1, 2021-05-04:
@OhPoorPup I do have a PS1D attached. I ran out of time to test the analog out to rule out PS1D.  Will try the suggested setting when I get back home and report my results.

ramapcsx2, 2021-05-04:
Please do, the game seems to work fine here :)

samson7point1, 2021-05-04:
Well the vertical and horizontal under scaler options are set to auto but they don’t seem to have a “0” setting - the settings are 1x, 2x and so on.  They don’t make this any better, but I was able to validate that analog out is not having this problem so while I don’t know a solution, the xstation isn’t the issue.

closed 2021-05-04 by samson7point1

OhPoorPup, 2021-05-04:
Hey! If you advance from "Auto" to 1 and then press left, you'll go back to 0, not back to Auto. At least that's how my PS1Digital behaves. Not the most intuitive....

samson7point1, 2021-05-04:
Thanks!  I just found that thread.  I was trying to do the adjustment under the scaler menu which also has an option called “vertical” that’s set to “auto”.  I’ve never been in the “capture adjust...” menu before, lol.  Set that value and all is well again.  I really love this community.  Thanks so much for going out of your way to find and link that forum post for me.  @OhPoorPup Your act of kindness really made my day:)

OhPoorPup, 2021-05-04:
No problem at all!! When I saw you'd closed the ticket I was like "waitwaitwait I gotta find the thread!!!" Glad you got it all working :)