x-station/xstation-issues #116: ./ now in directory after refresh game list on latest firmware, clogging up main navigation menu

xstationfan, 2021-05-08:
I think I was on 1.0.4 now on latest 1.2.0

added new stuff (have <150 folders w bin cue per Alphabetized parent folder), did refresh game list for the first Time in a while, now seeing ./ game name in the main menu as the first entries

I have to scroll through all of these to get to the ones that don't have that as prefix , 

look to 00xstation/gamelist.txt on SD card and it has /gamename.bin but also ./_gamename.bin as duplicate now underneath it. 

on a Mac for file transfer.. but this was not a problem with 1.0.4 refresh gamelist 

thanks huge fan of your work Robert. This and GBS-C contributions (even tho I don't have one of those).

ramapcsx2, 2021-05-08:
This is an issue with Mac's Finder placing metadata on external storage media.
It'll be fixed with the next update, due soon!
Until then, you could either delete this hidden folder on the sd card, or try the beta 1.3.0, shared on Discord:
https://discord.gg/Fc4XVcGB

ramapcsx2, 2021-05-08:
Here's an article on it, but I don't have a Mac to say how many of these folders are created:
https://www.techrepublic.com/article/how-to-disable-the-creation-of-dsstore-files-for-mac-users-folders/

ramapcsx2, 2021-05-09:
I'm closing this issue to clean up a little. It will be fixed soon with the next update :)

closed 2021-05-09 by ramapcsx2