x-station/xstation-issues #118: Please Insert Playstation CD-ROM.

thesunnymachine, 2021-05-21:
I have had my Xstation working for months now. A few weeks ago I installed the 1.20 update and noticed there was a new update.  

I tried booting and suddenly I am getting the "Please insert PLAYSTATION CD-ROM."  Message. 

Checked continuity on all solder points.  No issues. 

Tried 3 different SD cards, same issue. Removed and reseated ribbon cable and removed the Lazerbear mount, SD card direct. nothing works.  

Blue light on the Xstation.  New SDcards get 00xstation folder.

I did get it to boot into the Xstation once after seeing this message early on, but now I can't get passed it.  

I wasn't having any issues at all until I launched Chrono Cross with the 1.20 firmware.  The memcard pro is reading Chrono Cross.

ramapcsx2, 2021-05-21:
"Please insert PLAYSTATION CD-ROM" means the modchip functionality isn't working.
This is a very narrow problem: Check the solder point "3" is connected, as well as the points marked "*" are grounded:
https://user-images.githubusercontent.com/6278722/112033652-f08d7380-8b3d-11eb-9124-131df4cb646f.png

thesunnymachine, 2021-05-21:
> "Please insert PLAYSTATION CD-ROM" means the modchip functionality isn't working.
> This is a very narrow problem: Check the solder point "3" is connected, as well as the points marked "*" are [...]

I have checked all the points with a multimeter. 

They are all getting a signal through. 

I've been playing the system every week or so for months without issue until tonight when I went to update.  

I will reflow all the solder tomorrow and report back but all points are working according to the picture.

thesunnymachine, 2021-05-21:
BTW, I've have no issues with my install since Feb 20th when I first installed this.  Outside of user error with where files go on the SD card. 

I can't imagine the soldering suddenly went bad with the console just sitting there.  :( 

I have a feeling the mod chip is stuck. 

I'm doing an install next month for a friend.  I can try his board in my PS1 and see if it's works.   But I'm hoping reflowing the joints will do the trick.

ramapcsx2, 2021-05-21:
Yea, I realize that it's strange, but I think it might be an intermittent contact. The older firmware might have made it somehow, while the newer doesn't. (There are always small timing changes.)

Is the console a 5501 type?

thesunnymachine, 2021-05-21:
> Yea, I realize that it's strange, but I think it might be an intermittent contact. The older firmware might have made it somehow, while the newer doesn't. (There are always small timing changes.)
> 
> Is the console a 5501 type?

Yes it's 5501. 

I will definitely report back after I reflow the joints.

ramapcsx2, 2021-05-21:
If you need to use your console now though, you can use any way to unlock the Mechacon and boot software through an alternative starter.
The recent "tonyhax" for the Memory Card would work, or Unirom on an expansion ROM cart.

However you want to do it, boot the loader with the 1.2.0 update file in your 00xstation folder, and install it.
If the theory holds, the old boot behaviour will be restored.

Please let me know if you try that route and it works :)

thesunnymachine, 2021-05-21:
> If you need to use your console now though, you can use any way to unlock the Mechacon and boot software through an alternative starter.
> The recent "tonyhax" for the Memory Card would work, or [...]

I am able to get into the launcher using tonyhax.  Games load and run fine, but fresh boots still give me the insert disc screen. 

Would it be safe to update the firmware to 1.3 after getting into the menu with tonyhax?

Still think it's a weak solder joint?

thesunnymachine, 2021-05-21:
Full load doesn't work, but fast load does.

ramapcsx2, 2021-05-21:
Okay, it's again confirming that the unlock signal doesn't work.
Using the hack bypasses the need for the unlock, hence things work until a reset locks the system again.

Yes, I still think it is related to the soldering, somehow maybe connected to firmware timings as well.
It is safe to upgrade or downgrade.

thesunnymachine, 2021-05-21:
> Okay, it's again confirming that the unlock signal doesn't work.
> Using the hack bypasses the need for the unlock, hence things work until a reset locks the system again.
>
> Yes, I still think it is [...]

Okay, good news. 

I reflowed and added solder to every point.  

Boots normally now and I was able to update to 1.30.

Thanks so much for the help!

closed 2021-05-21 by thesunnymachine

ramapcsx2, 2021-05-21:
Awesome :)