x-station/xstation-issues #120: New install on PU-8. Everything working except the long reset to loader.

mnovoas, 2021-05-25:
Well, as stated in the title, only the long reset seems not to be working.
Did the install yesterday and tested a few games. Tried to reset to loader and it never worked, it always loads running game.
Also updated firmware and the console stopped at the screen mentioning that it would reboot automatically. It didn't, I rebooted it after about 20 minutes (just to be sure :))
Looks like the xstation isn't able to "order" the console to reset?
I've opened it up today and rechecked all the points according to the very helpfull install notes by mmmonkey (kudos to you, sir) and found no issues (from the board to the flatcable connector). 
Probably not related but also tested the drive closed switch for intermittencies, but no, it seems also to be working as expected.
Any ideas on further troubleshooting?
Thanks!

ramapcsx2, 2021-05-25:
Hey,
the xStation drags down the PSX reset line, forcing all the chips to reset.
If that reset low signal doesn't reach the chips, no reset can occur.
2 situations can cause this:
- the reset signal is not properly soldered down, and the multimeter test only succeeds because of the probe pressure in the measurement
- there is another device on the reset line that overpowers the xStation pulse

It is pin 17 in the install pics:
https://github.com/x-station/xstation-issues#late-pu-8--21-and-later

mnovoas, 2021-05-29:
Hi, I'm sorry for the delay, only found time today to look at this.

So, you're right, it was the first issue. Everything seemed ok with the board outside but somehow the tension of the flatcable was messing with that area of the qsb and I wasn't noticing that. Remade that pin and solved that problem.
Thanks for the help!

ramapcsx2, 2021-05-29:
Great! Glad it's been solved :)

closed 2021-05-29 by ramapcsx2