x-station/xstation-issues #122: NTSC games on PAL console

HE1NZ57, 2021-05-31:
Playing NTSC games on PAL consoles results in a rainbow colored image. Sound is garbled. PAL games work fine, but audio is also a little garbled.

I'm not sure if it's a bad install or some XStation issue. I googled it and some other guy had the same problem, he's been told the console can't play NTSC games? That can't be true because when modchipped the PAL PS1 runs NTSC games just fine at full 60hz speed.

If XStation indeed can't play NTSC games on PAL console then it's a huge issue that wasn't advertised, basically makes my purchase a giant waste of money.

I don't know where to ask question about this other than here.

ramapcsx2, 2021-05-31:
That's fine. It's a common problem.
The issue is that Playstation consoles before the SCPH-700x series do not output a correct color encoding frequency for out of region games.
It has nothing to do with the ODE. It is a system limitation.

What you want to do now is either:
- use an RGB cable, which bypasses the need for encoded video
- get something like a DFO mod installed, which corrects the frequencies for both modes
- get ps1hdmi ( :p )

HE1NZ57, 2021-05-31:
Thank you for answer.

I'll probably install DFO if I can find a cheap solution, because I already have Retrotink. It can't play RGB though and I shipped expensive original S-Video cables. You should probably tell that PAL consoles come with this issue. For a common problem it sure is very underreported. I've watched multiple reviews of the product and nobody ever mentioned this. It says "Automatic 50/60Hz switching" on Castlemania games, I assumed there won't be any issues. Especially since modchip does the same thing perfectly.

Can you comment about the sound? Is it faulty install? Can encoding frequency cause bad audio?

ramapcsx2, 2021-05-31:
There's always several aspects to this.
For one, I consider this such common knowledge, I always wonder when someone sees this for the first time.
Even in 2000, it was common knowledge (in my circle :) ) that you need an RGB cable for color, and a modchipped console came with the cable.

Next, what you describe as working with modchip has to be a later model console.
Those don't have the correct frequencies either, but the way they work enables something called PAL60, which most devices understand.
Unfortunately, this isn't available as an option with xStation, due to those consoles not having the DSP bus available.

Next, when people review the ODE, they are often based in the US, playing NTSC games on NTSC hardware.
Very rarely you'll find someone over there wanting to play a PAL title, but when they do, the same issue (in reverse) will come up.
You need an RGB cable to avoid all this.

So with all these variables, how should this be reported?
There'd be so many disclaimers required, people would probably be confused more than helped.

HE1NZ57, 2021-05-31:
The other console I have has the same 5502 motherboard. I don't know what kind of chip is installed, but I get good colors when I burn NTSC images. Maybe I got lucky and have a good chip.

You can add a notice that XStation doesn't make your console region free and additional modding required to play games from other regions.

I guess I'll send console back to the modder to check if sound issues are caused by bad install and maybe he can order that osscilator thingie.

ramapcsx2, 2021-05-31:
Well, there is no clean modification (short of a DFO) that enables a PU-18 to display both region colors correctly.
Whatever is in there must be providing a fixed color carrier clock. It's possibly a crude DFO-like modification that sometimes was done.

The xStation does make the console region free. This is part of what I meant with "adding to confusion" :)
What it doesn't do is fix the Composite Video / S-Video color encoding frequency, which is pretty specific.

Adding a DFO would be a good fix for your setup. Most others should go by the old rule of using an RGB cable.

inteyes, 2021-05-31:
I ended up just using the RGB SCART cable on my PAL console to play NTSC games.
In this case, no additional modification is required.

Regarding colors - I know that in our country most of the consoles were equipped with some kind of color mod in the 2000's, so that might explain why people could not even aware of this problem exists :)

HE1NZ57, 2021-06-19:
Is there a place that sells DFO mod specifically for PS1 systems? I'm struggling to find one. I managed to fix the audio issue, it was related to poor soldering.

ramapcsx2, 2021-06-19:
They pop up in small mod shops here and there, sometimes a larger store has them.
https://www.consolesunleashed.com/product/sony-playstation-dual-frequency-oscillator-mod-kit/

HE1NZ57, 2021-06-19:
> They pop up in small mod shops here and there, sometimes a larger store has them.
> https://www.consolesunleashed.com/product/sony-playstation-dual-frequency-oscillator-mod-kit/

Thanks. 

So, if I order these, install them on my PAL systems, it's going to automatically switch between PAL and NTSC and display all games properly, right? Description says something about composite signal. 

This guy seems like he might not be sending to Russia. I'll email him.

ramapcsx2, 2021-06-19:
Yes, though the wording is not quite correct. I think it's important to get this right.
Every PSX can switch between PAL and NTSC via a simple software command. This is why games run in 50 or 60 Hz, based on which region they were meant for.
What the PSX cannot do though, is switching in a correct color subcarrier clock for the out of region display mode.
The DFO provides this clock, and so restores color encoding for Composite and S-Video.

HE1NZ57, 2021-06-19:
Alright. Gonna order soon.